sudomesh/sudowrt-firmware #22: Consider running polipo

Juul, 2015-02-20:
polipo is used by both the EFF firmware and Open Garden to speed up the browsing experience. It really does work. We should consider setting it up. Here is the config from the EFF firmware:

  https://github.com/EFForg/OpenWireless/blob/master/etc/config/polipo

max-b, 2015-03-14:
What do we think about the node's ability to run this considering their software constrained speeds? Do you think that with the new hardware it won't really be a big deal?

max-b, 2015-03-14:
Well just installed polipo on one of the nodes in my house and it really does speed up web browsing (can't just be placebo can it?)!

I'll keep an eye on resource usages, but for now it looks fine.

max-b, 2015-03-15:
I added polipo to the packages list and a config to makenode
https://github.com/sudomesh/makenode/commit/77fb9b0171d335feb8bf9911cea32a48b1e11fae
https://github.com/sudomesh/sudowrt-firmware/commit/f5822c37a2b57b3558f8a9014456aa07497bc803

So far resource usages look ok on the node that has been running polipo:
http://192.241.217.196/cacti/graph_view.php?action=tree&tree_id=1&leaf_id=40&select_first=true

We can keep an eye on them.

closed 2016-06-17 by Juul