sudomesh/sudowrt-firmware #24: polipo and WPAD

Juul, 2015-03-10:
For nodes with enough ram and/or usb disk storage we should consider adding polipo and configuring proxy autoconfiguration (WPAD) using dnsmasq's DHCP and a web server. 

Polipo really speed's things up for connections with slow latencies so making it as easy as possible for people to set their browsers to use polipo is kinda great. Open Garden uses polipo, which I believe accounts for a large amount of the perceived performance improvement, but they use it as an intercepting proxy, whish is nice for most users, but bad in general, since it _forces_ port 80 traffic through a proxy (and a proxy that doesn't support e.g. webdav, causing all webdav requests to fail). 

Unfortunately we're doing away with the web server so it's kinda annoying we'd need one _just_ for this purpose.

Maybe polipo isn't worth it since everything's swtiching to https?

max-b, 2015-03-15:
Hmmm ok well I did add polipo to a node at my house and to the sudowrt configs see #22 . What sort of tests do you think I should run in order to test functionality?

paidforby, 2018-10-10:
polipo was added, but should now be removed #95

closed 2018-10-10 by paidforby