sudomesh/sudowrt-firmware #28: mDNS forwarding

Juul, 2015-06-26:
We need mDNS relaying for for DNS-SD (and thus the service browser) to work across the mesh.

David Taht has this to say:

> I shipped avahi as a reflector in cerowrt. It caused broadcast storms
> with more than three nodes interconnected. It also went nuts when a
> device was on more than one subnet and announced itself, doing things
> like renumbering nodea-1 to nodea-2,3,4,5,6 (I dont think avahi is RFC
> compliant here)
> 
> You are SOL.

from [this email](https://sudoroom.org/pipermail/mesh-dev/2015-June/000213.html)

We could try using [mdns-repeater](https://bitbucket.org/geekman/mdns-repeater) but we probably want to read the code and think about whether we might run into similar problems. We might want to build in some packet storm protection (though shouldn't Babel prevent loops?). We could cache a checksum of each forwarded mDNS packet for some number of seconds (with some sane maximum buffer size) and refuse to forward if a packet matches a recently forwarded packet.

max-b, 2015-07-06:
I spent a while thinking about mDNS stuff this morning and I'm feeling like there are some significant drawbacks that we're maybe not considering. How do we handle hosts on the network that aren't running zeroconf? Won't they just not be able to do hostname resolution? 

I brainstormed a bunch of alternatives, but I don't think that any of them would be feasible. The solution that I think would be the best - a combination/translation layer between dnsmasq and mdns is apparently not feasible:
http://lists.thekelleys.org.uk/pipermail/dnsmasq-discuss/2007q2/001356.html
http://lists.thekelleys.org.uk/pipermail/dnsmasq-discuss/2013q4/007754.html

This is sort of the broad outlines of what I was thinking, but I think I'd have to follow up a bunch more to see what exactly they ended up doing.

Juul, 2015-07-10:
I can think of a quick and dirty hack that will enable .local A record resolution via normal unicast DNS: Patch dnsmasq so A record queries for .local are resolved using gethostbyname, which on a system that has avahi will result in an mDNS lookup. We'd be constructing a fake DNS response packet for dnsmasq to cache, meaning we'd have to fake the TTL. Not great but if we stick to the avahi default for TTL it probably wouldn't be terrible. 

Or if we don't want to be hacky we could make a fork of dnsmasq that depends on avahi and simply use the avahi functions for .local lookups from within dnsmasq. The code to do a simple lookup is already available in the avahi source code in avahi-utils/avahi-resolve.c

Juul, 2015-07-10:
Of course the dnsmasq author would never do such a thing, but if we are already putting avahi and dnsmasq on all nodes there really isn't anything bad about having them depend on each other. The avahi code we need should be in the avahi-*.so shared libraries so we shouldn't have to sacrifice disk space.

Juul, 2015-07-10:
Someone also wrote this: 

https://github.com/pssc/dns-to-avahi

But that's definitely not something we'd want to run on nodes so then we're stuck having a few machines running DNS servers for the entire mesh (probably the same machines running service browser), and we're in the weird situation that a citywide mesh is relying solely on perl libraries for DNS resolution and caching.

max-b, 2016-01-14:
@Juul I'm wondering whether or not we'd rather use this:
https://github.com/mwarning/KadNode

It's gonna take a little bit of finagling I think to get it to play nicely with the rest of our dns setup, but I think it makes more sense than trying to extend mdns...