sudomesh/sudowrt-firmware #34: Complete new home-node web UI

Juul, 2015-06-26:
So we can finally ditch luci.

max-b, 2015-12-18:
I've got a working, basic implementation of the web ui: https://github.com/sudomesh/sudowrt-web-ui/commit/773f2e18ea90633191c9ff53dffb254f92089469

which can be installed by this package:
https://builds.sudomesh.org/builds/chaos_calmer/ar71xx/packages/peopleswifi/sudowrt-web-ui_0.0.1-HEAD_ar71xx.ipk

max-b, 2015-12-26:
Firmwares are now defaulting to build with `sudowrt-web-ui` and not luci. All of the features we had previously listed are available now: changing private ssid and password; amount of bandwidth to share; and admin password.

The web ui could use better feedback for things loading/saving.

max-b, 2016-01-26:
I'm gonna close this ticket as is and we can pick individual pieces of polishing the home node web ui and ticket them separately.

closed 2016-01-26 by max-b