sudomesh/sudowrt-firmware #37: Complete basic network security audit

Juul, 2015-06-26:
Ensure that there is no way to talk to anything on the WAN or private wifi networks from the open or mesh networks.

max-b, 2015-09-09:
@thebalaa said that at one point last night he connected to to the "peoplesopen.net" SSID wirelessly and his traffic was going directly out through his comcast connection - not being tunneled. This was on a tp-link wdr3500 (the n600).

 I'd like to try to replicate this before we start digging too deep into it as a bug. My understanding is that these lines: https://github.com/sudomesh/makenode/blob/master/configs/ar71xx/tp_link_n600/templates/files/etc/init.d/meshrouting#L110-L118

should prevent that behavior. That being said, if for some reason they're not working as expected, then we should certainly check it out.

max-b, 2015-09-09:
Just brainstorming that issue - I'm wondering if there's a timing problem and/or an init script dependency issue. For example, I don't really know what exactly would happen if someone ran `/etc/init.d/network restart` without also then running `/etc/init.d/meshrouting restart` Similarly, I wonder if there's a brief period of time after network restart where the meshrouting rules haven't been added yet?