sudomesh/sudowrt-firmware #38: Make extender-nodes also extend the private wifi network

Juul, 2015-06-26:
This is trivial to do. Opinions are split on this. Mitar thinks even having the private wifi network is not a great idea since it just uses mesh frequencies for people's private traffic. wrought made the point that it is a common use-case for people to want to share all of their internet bandwidth with someone they trust, e.g. their neighbors. We've actually already had one person request this feature. Splitting one expensive connection between two or three houses/apartments just makes sense and the bandwidth limit on that should not be the same as on the publicly shared connection, nor does it need a VPN. This is also a form of sharing that we want to encourage, even if it is much more localized than the usual "share bandwidth with everyone" scenario.

Juul, 2015-06-26:
This is trivial to implement. It's the exact same solution as for the open network.

eenblam, 2018-10-10:
Reasons I think this should be closed for the time being:
- Depending on the extender node configuration, this could be costly in terms of performance for relatively directional links, e.g. a sector antenna. If a user tries to connect from outside of its cone, this will degrade its overall performance. The user won't be happy with the resulting bandwidth, and it degrades the overall quality of the link as a result.
- We aren't actively developing for a target extender node platform, and LibreRouter is currently looking like the most promising option in the near future. Let me know if I'm wrong, but my understanding is that we'd probably be able to drop our home/extender dichotomy if we jumped for LibreRouters anyway.

paidforby, 2018-10-10:
Agreed @eenblam if we do go back to putting OpenWrt on extender nodes, we should drop the distinction between home nodes and extender nodes (i.e. any extender node, Ubquiti, Mikrotik, librerouter, etc. would be able to function as a node in its own right).

closed 2018-10-10 by paidforby