sudomesh/sudowrt-firmware #43: Automatically discover and select a VPuN (exit) server

Juul, 2015-06-26:
We need to be able to run multiple VPuN servers and have the clients discover them and try connecting to them one by one until one succeeds.

Juul, 2015-07-10:
Mitar pointed out that we don't want multiple exit servers though, since lots of stuff won't like it when the user switches IPs. If we get an AS number there are ways to deal with this.

paidforby, 2018-03-07:
this is fixed by patch for https://github.com/sudomesh/bugs/issues/17 closing duplicate

closed 2018-03-07 by paidforby

bennlich, 2018-03-07:
@paidforby Is a VPuN relay server the same thing as an exit node?

paidforby, 2018-03-07:
Yes, as far as I know. VPuN is a "Virtual Public Network" so essentially another term for the "Mesh Area Network" or the "network" created by tunneldigger and routed over by babel.

bennlich, 2018-03-07:
👍