sudomesh/sudowrt-firmware #44: Run OpenVPN on trusted servers

Juul, 2015-06-26:
We should look into this since most devices nowadays can be configured to auto-connect to an OpenVPN server on wifi connect.

A major concern for people is if other's can sniff their traffic while using the mesh. I've used the "it's the same as when in a cafe" explanation but apparently many people are concerned about using cafe wifi as well. 

The correct answer to the concern is "always use end-to-end encryption" but that's not really reasonable for the average person (yet). 

Having encryption on everything as an easily configured option, at least until reaching the internet, should assuage most concerns as long as only trusted individuals/organizations run these OpenVPN servers. Unfortunately this means that each device will be configured for _one_ OpenVPN server so if that server goes down peoplesopen.net will probably stop working for that device until the configuration is changed. This is essentially introducing a type of single point of failure.

paidforby, 2018-10-24:
@Juul any links or new ideas on how to actually make this happen, seems like a cool idea and would be make explaining mesh security easier. Or this irrelevant now that most sites use HTTPS or at least it is more visible to people.

muxlux, 2018-10-24:
could we use wireguard (kernel layer secure vpn) on nodes & exit nodes &
import node database?

-- 
*First they ignore you, then they laugh at you, then they fight you, then
you win*.
- Mohandas Karamchand Gandhi

CONFIDENTIALITY NOTICE:
The contents of this email message and any attachments are intended solely
for the addressee(s) and may contain confidential and/or privileged
information and may be legally protected from disclosure. If you are not
the intended recipient of this message or their agent, or if this message
has been addressed to you in error, please immediately alert the sender by
reply email and then delete this message and any attachments. If you are
not the intended recipient, you are hereby notified that any use,
dissemination, copying, or storage of this message or its attachments is
strictly prohibited.

eenblam, 2018-10-25:
We would have to roll our own key distribution to run Wireguard. Are you
thinking of using it in place of tunneldigger?

muxlux, 2018-10-25:
Are you thinking of using it in place of tunneldigger?
>
yes :)

paidforby, 2018-10-25:
Thanks @muxlux , wireguard looks cool, I was trying to remember the name of that.
@eenblam I don't think this would replace tunneldigger, necessarily. I would look at wireguard as potentially a secure methods of tunneling over babeld within the mesh itself. That seems to be the original topic of this issue.

Additionally, I'm cautiously in favor of dropping the tunneldigger idea all together. Say we got access to some amount of bandwidth in Oakland or Richmond, why would we need tunneldigger? I suppose tunneldigger could still be used at gateways that aren't directly connected to that bandwidth. But why not do that with something that would also work on the mesh and would (hopefully) not lock us in to a corner with IP addresses in config files, a la #148  and https://github.com/sudomesh/bugs/issues/28

Note, there's is also a Layer 3 tunneling protocol called fastd, mentioned in #41.

eenblam, 2018-10-25:
> and sudomesh/bugs#28 <https://github.com/sudomesh/bugs/issues/28>

@paidforby not suggesting it, just wasn’t sure of the scope of the
discussion.

Replacing tunneldigger with wireguard would mean arbitrary clients would
need to be able to securely provide a key to and receive credentials from
each exit node they access. We’d have to figure out our own PKI, since
wireguard leaves this entirely as the user’s problem to solve. Perhaps this
is fairly solvable, but I think this would make it much harder to support
our current “anyone can connect to the mesh” feature.

Curious about fastd, but know very little about it.

bennlich, 2018-10-26:
The wireguard docs seem well-written, and the conceptual model seems relatively simple, which is nice.

It's hard for me to tell how many people actually use it. Also there's this disclaimer:
```
WireGuard is not yet complete. You should not rely on this code. It has not undergone proper degrees
of security auditing and the protocol is still subject to change. We're working toward a stable 1.0
release, but that time has not yet come. There are experimental snapshots tagged with
"0.0.YYYYMMDD", but these should not be considered real releases and they may contain security
vulnerabilities (which would not be eligible for CVEs, since this is pre-release snapshot software).
If you are packaging WireGuard, you must keep up to date with the snapshots.
```

--

Associating a private key with each mesh ip in the meshnode db would be pretty straightforward.

We would need to start configuring nodes with exitnode public keys in addition to the exitnode ip addresses that are already a part of the node config files. @eenblam is there anything security-silly about simply adding these exitnode public keys to a new config file in the firmware?

--

> wait, wireguard just sets up a wg0 interface couldnt it run alongside whatever & just be an option til we flush out auto pairing ip w/key on node database

@muxlux yeah I think you're right

Its not yet clear to me if it would be easier to add wireguard on top of tunneldigger (as an optional encryption layer), or to use wireguard side-by-side with tunneldigger (i.e. as an optional replacement). I think I'm wondering which of those options requires the fewest / least intrusive changes to our current exitnode config. 

I'm currently thinking side-by-side is the way to go--I think wireguard can handle NAT traversal by itself: https://www.wireguard.com/quickstart/#nat-and-firewall-traversal-persistence.

paidforby, 2018-10-26:
@bennlich I'm wondering the same things regarding the best way to implement a secure tunneling protocol, whether it be fastd or wireguard or something else.

Originally, I was thinking that tunneldigger could not be run over babeld  because babeld relies on IPv6 multicast messages to perform neighbor discovery and tunneldigger does not support IPv6. However, we are running babeld on the tunnel broker to allow all nodes to virtually mesh, I'm not sure how the tunnel broker accomplishes this. 

Regardless of babelling abilities, L2TP does not provide any security. It can act as a secure VPN when paired with IPSec, but wlan-slovenja chose not to implement this in tunneldigger because,

> Confidentiality of communications was not a requirement, since the whole network is open anyway and openly transmitted over wireless, so we could further remove the overhead of cryptographic operations. 

This is valid, but security/privacy is always the first or second question people ask when they first hear about peoplesopen. It would be cool to implement a method of tunneling that was secure.

Given the above considerations, we have few potential options:

1. Side-by-side: Allow gateway nodes to choose whether to use a tunneldigger broker or secure tunnel server as their exit node.

2. On top: Leave tunneldigger broker/client setup as is. Nodes with internet connections act as "gateways" to the tunnel broker. Other nodes use secure tunneling over the mesh, to a gateway node, out of the L2TP tunnel, and to a VPN server.

3. Complete switch: Use only a secure tunneling protocol. Every node tunnels over every other node and eventually out to a VPN server. I'm curious how mesh routing would work in this setup.

4. Expand current functionality: Add an IPSec option to tunneldigger and maybe figure out how to get tunneldigger running over the mesh.

"Side-by-side" could be a simple addition, but in my experience, if you try to maintain two options, one will almost always end up being neglected.
"On top" would probably be more complicated, but could get us to testing tunneling over the mesh sooner.
"Complete switch" is more of the ideal scenario that we would be working toward by implementing one of the first two options.
"Expand current functionality" seems like a lot of work, though could be a cool addition to tunneldigger.

Would be interested to see someone get wireguard running on a real node!

Links of interest:

freifunk's gluon have something called Mesh-VPN,
https://gluon.readthedocs.io/en/v2018.1.x/features/vpn.html

using wireguard over babeld, a little sparse, but might have useful insights,
https://blog.fugoes.xyz/computer/network/2018/02/03/Run-Babeld-over-Wireguard.html)

a good breakdown of the reasoning behind tunneldigger,
https://wlan-si.net/en/blog/2012/10/29/tunneldigger-the-new-vpn-solution/

I also dug up some early sudomesh exploration of babeld, mDNS, and multicast layer 3 psuedo-bridges on an old email thread, seems relevant to this discussion
https://sudoroom.org/pipermail/mesh-dev/2015-June/000211.html
https://sudoroom.org/pipermail/mesh-dev/2015-June/000213.html

eenblam, 2018-10-26:
I feel like "side-by-side" wouldn't really address the privacy/security issues detailed in the thread, as @paidforby mentioned a few days ago.

If we're using, for example, Wireguard to secure traffic that is headed for an exit node, then that's awesome and a huge step forward. The "on top" option seems preferable to me in that regard. What about traffic on the mesh, though? Then each node needs key exchange with each other node it might need to send traffic through. For example, if Manatee wants to reach Dolphin through Shark, Manatee and Dolphin would have needed to exchange keys at some point. We could dodge this if all traffic on mesh first went through an exit node (and two Wireguard tunnels,) but I don't think anyone would be happy with that.

@muxlux @bennlich sorry I wasn't clear that this is what I meant earlier in the week about key exchange - I wanted a more general solution that also works on-mesh, not just protections for our exit node traffic. I wasn't taking enough to time to clarify.

If "add IPSec to tunneldigger," "figure out how to get tunneldigger running over the mesh", and "use only a secure tunneling protocol" are on the table, it sounds like we're moving towards tackling some new distributed systems problems. I'm excited to see where we go with this.

bennlich, 2018-10-28:
> If we're using, for example, Wireguard to secure traffic that is headed for an exit node, then that's awesome and a huge step forward. The "on top" option seems preferable to me in that regard.

Hmm not sure I follow why you would prefer  "on top". Couldn't you secure traffic to an exitnode with a wireguard tunnel by itself? Could even remove tunneldigger altogether or leave it for use by nodes that haven't performed a key exchange.

> What about traffic on the mesh, though? Then each node needs key exchange with each other node it might need to send traffic through.

Yesssss. Ambitious! But seems cool and doable. I think it would be more accurate to say: "needs key exchange with each other node it might need to send traffic **to**" (instead of through). E.g. you and I can exchange keys on paper and have secure communication with each other across the mesh forever after.

Maybe that can be v0 of our key exchange protocol--bring a slip of paper to the next BYOI (am I channeling @jhpoelen?)

One nice thing is that securing connections to the exitnode and securing connections between mesh nodes are somewhat orthogonal issues. Don't need to solve both at once.

eenblam, 2018-11-07:
> Hmm not sure I follow why you would prefer "on top".
@bennlich Well, one big reason would be backwards-compatibility.

Also, there's been a recent discussion about using Wireguard for a mesh in the Wireguard mailing list.

https://lists.zx2c4.com/pipermail/wireguard/2018-November/003536.html

closed 2018-11-07 by eenblam