sudomesh/sudowrt-firmware #45: Merge newest Babel 1.6.1 changes into our Babel version

Juul, 2015-06-26:
We're currently at 1.5.1

max-b, 2015-12-15:
I had to do this by cherry-picking the relevant commits from sudomesh/babeld/master onto jech/babeld/master:

https://github.com/sudomesh/babeld/tree/maxb/upstream-fixes

I squashed it down to just one commit:
https://github.com/sudomesh/babeld/commit/4a4c1224984d9ce5a18da43d99771970e67145ef

The only catch is that we're probably going to have to re-write history on the sudomesh/babeld/master branch. I don't think it'll be a problem, but I'm going to hold off on doing it for a few.

eenblam, 2018-06-10:
@Juul Can we close this in favor of working to move back to jech's babeld?

bennlich, 2018-06-10:
I'm not @Juul , but am comfortable w/ yes :)

closed 2018-06-10 by bennlich