sudomesh/sudowrt-firmware #46: Support Picostation, Picostation M, Bullet and Meraki Outdoor (sparky) as extender nodes

Juul, 2015-06-26:
The Picostation, Bullet and Meraki we want to support because we have a ton of them (and they're cheap).

The Picostation M because they're a decent choice as outdoor extender nodes.

Juul, 2015-07-01:
cpuinfo from Nanostation M2:

```
root@OpenWrt:~# cat /proc/cpuinfo 
system type		: Atheros AR7241 rev 1
machine			: Ubiquiti Nanostation M
```
Be aware that there are [two versions of the Nanostation M5](http://wiki.openwrt.org/toh/ubiquiti/nanostationm5) (old ar724x and new ar9342) and that the new one needs the -xw openwrt firmware image. This is not true for Nanostation M2.

Juul, 2015-07-01:
There are two ports on the Nanostation M2, the Main and the Secondary. It also has a built-in switch:

```
root@OpenWrt:~# swconfig dev switch0 show
Global attributes:
	enable_vlan: 0
Port 0:
	pvid: 0
	link: port:0 link:up speed:1000baseT full-duplex txflow rxflow 
Port 1:
	pvid: 0
	link: port:1 link:up speed:100baseT full-duplex auto
Port 2:
	pvid: 0
	link: port:2 link:down
Port 3:
	pvid: 0
	link: port:3 link:down
Port 4:
	pvid: 0
	link: port:4 link:down
VLAN 0:
	vid: 0
	ports: 0 1 2 3 4 
```

but this switch is wired differently from the built-in switch in e.g. the TP-Link N600 routers. The VLAN functionality is disabled and switch port 0 and 1 map directly to the eth0 and eth1 interfaces, which are labeled as Main and Secondary on the device:

```
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default 
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP group default qlen 1000
    link/ether 68:72:51:25:90:5e brd ff:ff:ff:ff:ff:ff
    inet 192.168.1.1/24 brd 192.168.1.255 scope global eth0
       valid_lft forever preferred_lft forever
3: eth1: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP group default qlen 1000
    link/ether 6a:72:51:25:90:5e brd ff:ff:ff:ff:ff:ff
4: wlan0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether 68:72:51:24:90:5e brd ff:ff:ff:ff:ff:ff
```

I have verified that creating VLANs on these ports using the normal method:

```
ip link add link eth0 eth0.1 type vlan id 1
ip link set eth0.1 up
ip addr add 10.0.0.1 dev eth0.1
# assume 10.0.0.2 exists on VLAN 1
ping 10.0.0.2
```

This means that we can ignore the switch completely and that we won't have to do any special hook script configuration for these devices. We can come back and figure out what to do with the secondary port at a later time.

Juul, 2015-07-01:
The Picostation M2 uses the bullet image.

It has no built-in switch and /proc/cpuinfo  says:

```
system type		: Atheros AR7241 rev 1
machine			: Ubiquiti Bullet M
processor		: 0
cpu model		: MIPS 24Kc V7.4
BogoMIPS		: 259.27
wait instruction	: yes
microsecond timers	: yes
tlb_entries		: 16
extra interrupt vector	: yes
hardware watchpoint	: yes, count: 4, address/irw mask: [0x0000, 0x0fe8, 0x0b68, 0x0bd8]
isa			: mips1 mips2 mips32r1 mips32r2
ASEs implemented	: mips16
shadow register sets	: 1
kscratch registers	: 0
core			: 0
VCED exceptions		: not available
VCEI exceptions		: not available
```

so for extender-node purposes it should be the exact same as the bullet.

The nanobridge also uses this image but I believe the nanobridge has two output antennas (two polarizations) so we'll have to account for that somehow.

max-b, 2015-09-07:
I have a working nanostation m2 and I've added the proper configs to the firmware builder here: 5d902ee4a8a1995deb43a352d7533788af215aa0

I also added a debugging interface to the nanostations second port in b432cf2, so that you can connect via that interface in case of issues. It's 172.22.1.2/24

@Juul - I'm confused about the picostations though - I thought they used the "atheros" firmwares as opposed to the "ar71xx"? To be perfectly honest, I never 100% understood the difference between those.

Juul, 2015-09-07:
On Mon, Sep 7, 2015 at 1:46 PM, max-b <notifications@github.com> wrote:

> I have a working nanostation m2 and I've added the proper configs to the
> firmware builder here: 5d902ee
> <https://github.com/sudomesh/sudowrt-firmware/commit/5d902ee4a8a1995deb43a352d7533788af215aa0>
>
> I also added a debugging interface to the nanostations second port in
> b432cf2
> <https://github.com/sudomesh/sudowrt-firmware/commit/b432cf288aaf33589f131ddd83f2de2b64a4bc77>,
> so that you can connect via that interface in case of issues. It's
> 172.22.1.2/24
>
> @Juul <https://github.com/Juul> - I'm confused about the picostations
> though - I thought they used the "atheros" firmwares as opposed to the
> "ar71xx"? To be perfectly honest, I never 100% understood the difference
> between those.
>
There are both picostation M2 which uses ar71xx and the old picostations
which use atheros.
ar71xx and atheros are different chipset families both made by atheros. The
one named "atheros" is obviously poorly named but was named "atheros" back
when they only had one chipset family.

-- 
marc/juul

max-b, 2015-09-07:
ah of course. Thanks for reminding me of that - for some reason it had slipped my mind...

Anyhow, that means that if we want to be able to use any of the remaining picostation 2s we have from the big ebay purchase we'll need to compile an `atheros` firmware. I'll look into that, but I don't think it will be too much different.

Juul, 2015-09-07:
On Mon, Sep 7, 2015 at 3:20 PM, max-b <notifications@github.com> wrote:

> ah of course. Thanks for reminding me of that - for some reason it had
> slipped my mind...
>
> Anyhow, that means that if we want to be able to use any of the remaining
> picostation 2s we have from the [...]
only difference should be the wifi config (802.11g only) and lack of
watchdog (which we aren't currently using anyway).

-- 
marc/juul

max-b, 2015-10-21:
Ok I looked into this and it turns out that openwrt has renamed the "atheros" chipset to "ath25": https://downloads.openwrt.org/chaos_calmer/15.05/ath25/generic/.

It should be an almost identical build process, so I'll work on implementing it.

max-b, 2016-01-11:
I have ath25 builds for the picostation 2, but not the bullets and nanostations (they have less flash):
https://builds.sudomesh.org/builds/chaos_calmer/ath25/openwrt-ath25-ubnt2-pico2-squashfs.bin

I can look into what might need trimming in order to fit on the bullets and nanos, but for now we at least have a firmware for the huge number of picostation 2s that we have. They'll be a good complement to the mynet n600s because they're slow but they have good range.