sudomesh/sudowrt-firmware #47: Testing: Compile, flash and configure a home + extender node set

Juul, 2015-06-26:
We need to ensure that everything comes up as expected

Juul, 2015-07-01:
Started compile of home-node firmware on room.sudoroom.org

```
tail -f /home/juul/sudowrt-firmware/built_firmware/builder.ar71xx/build.log
```

max-b, 2015-09-07:
The builds (mostly notdhcp stuff) have changed pretty significantly since Jul 1. I believe we're ready to do another full re-build of the tp-link n600 and n750 and the mynet n750 (I don't think I've finished the mynet n600 configs yet). As for the extender nodes, it looks like we should have support for the bullet m2 and m5 as well as the nanostation m2. I *think* the nanostation m5 should be a quick port of the m2 configs, but haven't attempted it yet.

As for the build server - isn't it also the sudoroom NAT exit server? Is it going to affect traffic on sudoroom's network if we're doing full re-builds of the firmware on it? I know that we're a little limited on options, but I was just curious if you thought it would have any effect.

Juul, 2015-09-07:
On Mon, Sep 7, 2015 at 1:51 PM, max-b <notifications@github.com> wrote:

> The builds (mostly notdhcp stuff) have changed pretty significantly since
> Jul 1. I believe we're ready to do another full re-build of the tp-link
> n600 and n750 and the mynet n750 (I don't think I've [...]
We have almost 100mbit of downstream so it should be fine. It should be
fast enough to handle traffic and compile. I tried to set up a machine as
mesh build server with a script to remote-wake-on-lan so we could wake it
up, rebuild and shut it down but I couldn't get it to wake on LAN. We can
still solder in a USB arduino to remote-boot via the sudo gateway i just
haven't done it yet.

-- 
marc/juul

max-b, 2015-10-05:
I've been playing with ansible and getting an more automatic build running on a vps server. I don't think it should be a permanent solution by any means, but it's been really helpful for testing, etc.

Right now I've got a working set of homenode and extender firmwares working with mynet n750 and n600 and tplink wdr4300 and wdr3600 (I think the wdr3500 works but I'm not 100% sure and I need one to test on).

The firmware builds are here: http://build-test2.sudomesh.org/builds/

I'm currently using makenode @ b25e0cc9778aad0a407cbce8938cfbb7599c1d4e, but I'll try to be careful about keeping it backwards compatible with the firmwares on that server.

max-b, 2015-10-05:
Oh and I just flashed and successfully got a nanostation m2 and m5 working as well as a bullet m2 and m5

max-b, 2015-11-03:
I'm keeping up to date builds on builds.sudomesh.org/builds.

It should be considered "bleeding edge" and we don't have anything that can be called "stable", but this is better than nothing and I tend to re-build new firmwares after significant changes/fixes that we've made.

closed 2015-11-03 by max-b