sudomesh/sudowrt-firmware #48: Hardware watchdog support

Juul, 2015-07-02:
ar71xx and newer supposedly have hardware watchdogs so we should use them

max-b, 2015-10-05:
I think it's already running by default:
```
root@my:~# ubus call system watchdog
{
        "status": "running",
        "timeout": 30,
        "frequency": 5
}
```

Juul, 2015-11-25:
Ok we should figure out how to disable that. I started implementing [this](https://github.com/sudomesh/watchpuppy) instead.

eenblam, 2018-06-10:
Please see [watchpuppy](github.com/sudomesh/watchpuppy) for @juul's solution. Please see [monitor](https://github.com/sudomesh/monitor) for our current monitoring infrastructure.

closed 2018-06-10 by eenblam

bennlich, 2018-06-13:
Hmmmm. I ran the command @max-b mentioned on my node, and got the same result, so I guess it's still running on our nodes.

@Juul why did you want to disable and replace with the puppy?