sudomesh/sudowrt-firmware #49: Manually specify frequency for Babel channel diversity

Juul, 2015-07-03:
Currently Babel isn't able to figure out the freaquency of wifi radios (this will be fixed at some point) so we need to set the frequency in the babeld conf ourselves

max-b, 2015-07-08:
Ok I've got this in 334613a 

This ends up being relevant to https://github.com/sudomesh/sudowrt-firmware/issues/51, so we should make sure we get it all incorporated.

max-b, 2015-07-08:
Oh and also of course we're gonna need to eventually get the other hosts, including the mynets and the extender nodes to incorporate the same.

closed 2015-07-08 by max-b

Juul, 2015-07-08:
Yeah. I can add it to the extender nodes. This will also need to be incorporated into the web UI so that changing frequency on e.g. an extender node also changes the babel frequency setting

max-b, 2015-07-08:
I added it to extender nodes: 0cb3525520dba710237b973f5214c6ef244c5436