sudomesh/sudowrt-firmware #50: Diagram and docs for extender node VLANs

Juul, 2015-07-03:
This really needs a diagram and an explanation

Juul, 2015-07-08:
Created a first shitty diagram [here](https://i.imgur.com/B0jzM9h.jpg). It's correct but looks terrible and needs an explanation.

paidforby, 2018-10-10:
This has been documented here https://github.com/sudomesh/babeld-lab/blob/master/operator_manual.md#mesh-network-components
The dashboard also has a diagram now, https://github.com/sudomesh/peoplesopen-dash#-1

closed 2018-10-10 by paidforby

bennlich, 2018-10-22:
We should put Juul's diagram above somewhere in the wiki if it isn't already there. Also probably would be a good exercise to try to redraw it by simply looking at the /etc/config/network and /etc/config/wireless files.

jnny, 2018-10-22:
The dashboard diagram is on the Walkthrough page - though it's technically incorrect as ports 1 and 4 are for extender nodes, 2 for public and 3 for private. Thought I'd uploaded the new version I made for the video:

[image]

Will update the wiki's /WalkThrough and /Technical pages with corrected diagrams - suggest we look through our Github docs as well to correct both the port connections and references to makenode..