sudomesh/sudowrt-firmware #55: Some way to update node firmwares

Juul, 2015-07-10:
gluon has a (centralized) update system that we could borrow, though we'd really prefer a decentralized system in the long run

jhpoelen, 2018-03-07:
Related see https://github.com/sudomesh/bugs/issues/18#issuecomment-368263669 .

gobengo, 2018-03-12:
My relevant thoughts on this are here: https://github.com/sudomesh/bugs/issues/20#issuecomment-369080052

In thinking through an implementation plan, I have started to think that maybe we are outgrowing just openwrt/buildroot for image building. We need something that builds not just a flashable image, but a 'package feed' of package updates, or maybe even something that can do over-the-air updates of the firmware image itself (nice because no 'configuration drift' of different combinations of packages).

I have been thinking that it might be ideal long-term to really define what each type of 'node' needs such that alternative build systems can be used to make 'valid/compatible' home nodes.

Personally I would like use this as a learning exercise for the Yocto Project, which is the Linux Foundation's cross-compiling toolchain for embedded linux development. I just took a training course on it at http://www.socallinuxexpo.org/scale/16x . Some advantages I see:
* Supports building packages/kernel for a ton of different chips
* Outputs a package feed for each package we have (e.g. peoplesopen-dash) of e.g. .rpms and/or .debs and/or .ipks in a way that is aligned with broader embedded Linux ecosystem. 
* Existing immutable update plugins like https://mender.io/

gobengo, 2018-03-14:
Relevant: https://github.com/theupdateframework/specification/blob/master/tuf-spec.md