sudomesh/sudowrt-firmware #56: Node monitoring system

Juul, 2015-07-10:
We need something. Someone who's not me probably has some ideas about what we could use.

max-b, 2015-07-10:
I have the most basic monitoring at http://192.241.217.196/smokeping/smokeping.cgi and http://192.241.217.196/cacti/graph_view.php (username is admin and password is the usual). All of the home nodes are shipping with snmp available with a common v1 string. Which means that the information is being transmitted in plain text. Ideally, we'd want to look into snmpv3 which supports encryption, but I'm not sure how well the most recent openwrt package version supports it. Also I don't know if the encryption will slow down overall performance.

On top of that - I hate cacti. Smokeping is simple, fairly easy to programatically write new hosts to, and provides a pretty useful metric. Cacti on the other hand, requires way too much hands on gui work and is touch at best. I've been looking into some collectd + influxdb + grafana setups (which is the new new hotness), but they're pretty beta, so it might take a little while to hack them into place. 

Is the meshnode-db currently up and query-able?

max-b, 2015-09-07:
meshnode-db is in fact up and no UI, but info is available.

Cacti now has publicly available graphs:
http://192.241.217.196/cacti/graph_view.php?action=tree&tree_id=1&leaf_id=45&select_first=true

Admin password has changed - ping me if you'd like it.

max-b, 2015-09-14:
Closing since we have basic monitoring and with #68 we'll be ok for milestone 0.2. If we want specific enhancements we can ticket them separately.

closed 2015-09-14 by max-b