sudomesh/sudowrt-firmware #60: sleep in nothdcpclient hook not long enough?

Juul, 2015-07-27:
I observed that an extender node did not have its IPv4 IP set on the mesh0 interface. Manually setting it using the line from the hook script worked fine. I expect that the interface was not yet ready. I'm increasing the sleep after the "wifi" command to 10 seconds.

Juul, 2015-08-04:
I now increased it to 30 seconds since 10 seconds did now seem to be enough. We definitely need something that polls until all wifi interfaces are ready and then continues.

max-b, 2015-09-07:
I also had to add a sleep line to the hook before the `brctl addif` lines because the bridge hadn't been created yet and it was silently failing by not adding the interface to the bridge.

max-b, 2015-10-02:
As for polling, ubus has: `wait_for <object> [<object>...]	Wait for multiple objects to appear on ubus` 

Also, the functions in `/lib/wifi/mac80211.sh` might be helpful.

max-b, 2015-10-02:
Ok it looks like this polling is working just fine now: 1ede2f713b4f6bc329f46edbfdc1271c22df6c6b

max-b, 2015-10-02:
Nope I don't believe it's actually working. Too bad - that was a particularly convenient looking fix....

max-b, 2015-10-02:
I think that the `waitForWifi` function that I wrote here 48a2d52eda471637a5b236e83fc0c80acc92ae18 is working. I suspect we won't need any of the other sleeps. I also want to look into `/etc/init.d/meshrouting` on the home nodes and see if that's where the sleep thing is happening.

max-b, 2015-10-18:
For future reference the waitForWifi function has been moved to https://github.com/sudomesh/openwrt-firmware-config/blob/master/files/usr/share/sudomesh/wireless.sh which is the "openwrt-firmware-config" package as marked in the "sudowrt-packages" feed. Consolidating/renaming is probably worthwhile.

closed 2015-10-18 by max-b