sudomesh/sudowrt-firmware #61: 172.22.0.2/32 is being broadcasted over babel

max-b, 2015-07-31:
The neighbors list on the exit server reads:
```
My id 06:01:40:ff:fe:fc:42:01 seqno 64286
Neighbour fe80::74f5:d0ff:fe50:997d dev l2tp2841 reach ffff rxcost 96 txcost 96 rtt 0.000 rttcost 0 chan -2.
Neighbour fe80::c4f6:b0ff:feb6:6a61 dev l2tp2831 reach ffff rxcost 96 txcost 96 rtt 0.000 rttcost 0 chan -2.
Neighbour fe80::9406:26ff:fe1d:d283 dev l2tp2811 reach ffff rxcost 96 txcost 96 rtt 0.000 rttcost 0 chan -2.
Neighbour fe80::ac15:2eff:fe0e:df08 dev l2tp1001 reach ffff rxcost 96 txcost 96 rtt 0.000 rttcost 0 chan -2.
100.64.0.1/32 metric 0 (exported)
0.0.0.0/0 metric 0 (exported)
100.64.0.10/32 metric 96 (96) refmetric 0 id 06:01:29:ff:fe:19:66:01 seqno 43192 age 2 via l2tp1001 neigh fe80::ac15:2eff:fe0e:df08 nexthop 100.64.0.10 (installed)
100.64.2.64/26 metric 224 (224) refmetric 128 id 16:cc:20:ff:fe:b5:5c:24 seqno 13513 age 10 via l2tp2811 neigh fe80::9406:26ff:fe1d:d283 nexthop 100.64.2.65 (installed)
100.64.2.66/32 metric 192 (192) refmetric 96 id 26:a4:3c:ff:fe:bd:29:c7 seqno 57727 age 10 via l2tp2811 neigh fe80::9406:26ff:fe1d:d283 nexthop 100.64.2.65 (installed)
100.64.2.192/26 metric 480 (480) refmetric 384 id 02:90:a9:ff:fe:0d:87:a3 seqno 38233 chan (255) age 4 via l2tp2841 neigh fe80::74f5:d0ff:fe50:997d nexthop 100.64.3.1 (installed)
100.64.3.0/26 metric 224 (224) refmetric 128 id 02:90:a9:ff:fe:0b:73:cd seqno 9974 age 4 via l2tp2841 neigh fe80::74f5:d0ff:fe50:997d nexthop 100.64.3.1 (installed)
100.64.3.64/26 metric 224 (224) refmetric 128 id 16:cc:20:ff:fe:75:c4:07 seqno 8334 age 7 via l2tp2831 neigh fe80::c4f6:b0ff:feb6:6a61 nexthop 100.64.3.65 (installed)
100.64.3.192/26 metric 480 (480) refmetric 384 id c2:4a:00:ff:fe:e4:8d:8f seqno 46937 chan (255) age 4 via l2tp2841 neigh fe80::74f5:d0ff:fe50:997d nexthop 100.64.3.1 (installed)
172.22.0.2/32 metric 192 (192) refmetric 96 id 26:a4:3c:ff:fe:bd:29:c7 seqno 57727 age 10 via l2tp2811 neigh fe80::9406:26ff:fe1d:d283 nexthop 100.64.2.65 (installed)
```

the next hop is 100.64.2.65 - is that a node at Pete's house? 

We should probably create more strict rules for the routes that certain nodes will install (*especially* the exit server....)

max-b, 2015-07-31:
Ok well I have good news and I have bad news.....

The good news is that I think I figured out what's causing this.

The bad news is that once again, it seems to be mtu issues :/  The home node had an mtu of 1446, while the exit server had an mtu of 1438. Once again, I really have no idea how these are both getting determined and then set. If I change the home node's mtu to 1438 to match the exit server, all of a sudden there are no more issues getting ssh connections across the link. 

I know that we can figure out what the hell tunneldigger is doing if we go over the code thoroughly enough. Considering it's such an integral part of our project at the moment, I think we may have to just bite that bullet... @papazoga - sound fun? :P

Juul, 2015-08-04:
yes we should figure out how to tell babeld to not install any non 100.64 routes

The babeld configuration of the extender nodes is just generally incomplete. We need to explicitly set the frequency for the wireless interface on first boot and we need to be explicit about which routes to redistribute. Finally we need to set the wired interface dynamically from the notdhcpclient hook script using uci since it is right now hard-coded to eth0.1

max-b, 2015-09-07:
> yes we should figure out how to tell babeld to not install any non 100.64 routes
Well except for the default route, which would be via the exit server.

I made a quick fix to this by adding `in ip 172.16.0.0/12 deny` to the exit server's babeld.conf file. But you're right that we should do a little more considering of babeld configs especially on the extender nodes.

max-b, 2015-10-02:
The issue about broadcasting the private ip range has been fixed in 800160dc775bea58cd95ff472059256635c566cd. 

 > We need to set the wired interface dynamically from the notdhcpclient hook script using uci since it is right now hard-coded to eth0.1
That is also done
> We need to explicitly set the frequency for the wireless interface on first boot and we need to be explicit about which routes to redistribute.
703fd81cc3383ef9dc3ff633afabb5ee5222d6f4 has the notdhcpclient up hook script set babel's interface channel :)
>  we should figure out how to tell babeld to not install any non 100.64 routes
I'll file a separate ticket for that because it's gonna be satisfying to close this one

closed 2015-10-02 by max-b