sudomesh/sudowrt-firmware #66: Give out mesh subnet route with dhcp?

max-b, 2015-08-20:
Currently when the homenode gives out an address, the client will receive that address and subnet, and will add that /26 subnet to its routing table with "proto kernel".

I believe we could configure the homenode to also give out the route for the entire mesh subnet: `100.64.0.0/10`

I think there might be pros and cons to this approach, but one pro that I see is that it will be an easy option for someone to be attached to a non-mesh connection over one interface and a mesh connection over another interface. We may want to keep the non-mesh default route, but also be able to connect to all of the other mesh services over the mesh route. This would normally be a simple process for someone using linux: `ip route add ...`, but we may end up often having to help folks setup osx or windows machines where it's much much less straightforward.

Just thinking out loud here for the moment....

Juul, 2015-08-20:
On Wed, Aug 19, 2015 at 8:53 PM, max-b <notifications@github.com> wrote:

> Currently when the homenode gives out an address, the client will receive
> that address and subnet, and will add that /26 subnet to its routing table
> with "proto kernel".
>
> I believe we could [...]

Huh? I'm not sure I understand. Why would that be a problem when using /26
vs. /10? If /10 is being handed out to clients then they will assume that
they don't need to go through the home node as a gateway in order to access
the rest of the /10, which is incorrect. As I understand it, the
expectation is that you are on the same layer 2 network as the rest of your
subnet so clients should get /26 not /10. Or am I missing something?

-- 
marc/juul

max-b, 2015-08-20:
I'm not saying that we should hand out /10 addresses. I'm saying that we should use the dhcp-option 33: http://www.networksorcery.com/enp/protocol/bootp/option033.htm to hand out a *route* for the entire /10 address space.

I have no idea which clients would support it....

Juul, 2015-08-20:
On Wed, Aug 19, 2015 at 11:41 PM, max-b <notifications@github.com> wrote:

> I'm not saying that we should hand out /10 addresses. I'm saying that we
> should use the dhcp-option 33:
> http://www.networksorcery.com/enp/protocol/bootp/option033.htm to hand
> out a *route* for the entire /10 address space.
>
> I have no idea which clients would support it....
>

Ok, but I'm still not seeing what it would accomplish? Maybe I just need
sleep.

-- 
marc/juul

max-b, 2015-08-20:
Somebody has a windows pc. They're connected over wifi to their normal home internet connection. Then they plug in an ethernet cable from their computer to one of the "open" (non-extender) lan ports. They'll get a lease on the mesh, but they won't have routes to anything beyond the /26 subnet they've been assigned in. Ideally we'd want them to have a route for the entire /10 mesh subnet that goes through that mesh interface.

Juul, 2015-08-20:
On Thu, Aug 20, 2015 at 11:46 AM, max-b <notifications@github.com> wrote:

> Somebody has a windows pc. They're connected over wifi to their normal
> home internet connection. Then they plug in an ethernet cable from their
> computer to one of the "open" (non-extender) lan [...]
The reason they wont have routes is because they now have two default
gateways and they won't know which one to pick? And you want to add a route
to the /10 (that specifies it has to go via the home node)? That makes
sense.

-- 
marc/juul

max-b, 2015-08-21:
Yeah I don't imagine that any OS does any sort of intelligent route sorting. I think they just take the first default route that's handed to them.

I'm test it out on my home node router and see how some different clients respond.

max-b, 2015-08-21:
Well it looks like this should be reasonably possible, but I'm not sure there's enough client support for it to warrant bothering with it at the moment:

We'd want to use option 121: "Classless Static Route Option" as articulated in https://tools.ietf.org/html/rfc3442
https://www.debian-administration.org/article/471/Supplying_routing_information_using_DHCP