sudomesh/sudowrt-firmware #73: Choose hostname

max-b, 2015-09-29:
Currently we use my.node for the hostname for every home node. I think we may want to have meaningful hostnames. I think it will make certain things related to administration and monitoring significantly more clear...

Juul, 2015-09-30:
On Tue, Sep 29, 2015 at 4:32 PM, max-b <notifications@github.com> wrote:

> Currently we use my.node for the hostname for every home node. I think we
> may want to have meaningful hostnames. I think it will make certain things
> related to administration and monitoring significantly more clear...
>
There is no reason we can't have meaningful hostnames. Each node should
have a unique hostname + a hostname that is always the same such as:
my.peoplesopen.net

The reason not to use my.node or similar is that most browsers will execute
a search for non-standard (and some standard) TLDs unless http(s):// is
specified.

max-b, 2015-10-05:
Ok what do you think of https://github.com/sudomesh/makenode/commit/7e1f0be109ca54de5e6b0064f02932c0e52b7e16 @Juul ?

max-b, 2015-11-03:
ok well a month later I'm fine calling this closed

closed 2015-11-03 by max-b