sudomesh/sudowrt-firmware #80: More flexible traffic shaping

max-b, 2015-10-19:
Depending on what kind of feedback we get about traffic shaping, we could implement a more flexible traffic shaping system that allows for bursts at higher rates than the ceiling for short periods of time. 

This is usually done with tc "buckets" http://linux.die.net/man/8/tc-tbf and I think it would probably allow for more smooth web browsing on peoplesopen connections that more strictly rate limited.

Juul, 2015-10-20:
On Mon, Oct 19, 2015 at 4:05 PM, max-b <notifications@github.com> wrote:

> Depending on what kind of feedback we get about traffic shaping, we could
> implement a more flexible traffic shaping system that allows for bursts at
> higher rates than the ceiling for short periods of time.
>

Cool. The danger is that the bursting can interfere with sensitive
communications like video conferencing so as long as we keep this in mind.
It might also be cool to include guides in the UI for helping people set
limits, e.g. "video conferencing takes about x kbits/sec per participant"
and maybe warn people if they're trying to share more than what's needed
for 100% always reliable one-on-one video conferencing.

-- 
marc/juul

max-b, 2016-01-11:
I honestly think that we should be ok now that we've incorporated the simple htb and fq_codel lines into our home node and tunneldigger implementations. 

Perhaps coming up with some very sophisticated traffic control rules might help us in the future as we're trying to optimize some links, but for now I think we're ok.

closed 2016-01-11 by max-b