sudomesh/sudowrt-firmware #84: Setup sudoroom to route "peoplesopen.net" traffic through VPuN tunnel

max-b, 2015-12-04:
The sudoroom gateway server is assigning addresses on the peoplesopen.net subnet, but it's not connected to our mesh tunnel gateway, so we can't access mesh services through it. 

I wrote some simple ansible playbooks to compile and run tunneldigger as a client on a debian server here: https://github.com/max-b/mesh-playbooks

I'm mostly just scared of some of the folks at sudo being pissed if the internet goes down, so I'd prefer that we at least have 2 people working on this so that we can check eachother's work...

eenblam, 2018-06-10:
Very fixed. Proof: https://peoplesopen.net/monitor

See also: https://github.com/sudomesh/exitnode

closed 2018-06-10 by eenblam

bennlich, 2018-06-13:
@eenblam Are you sure https://peoplesopen.net/monitor is proof of this? I think this issue is about whether the sudoroom gateway machine (which I guess lives in the cage?) is able to route traffic to the mesh intranet. I don't really know much about how sudoroom internet is set up though, and it may have changed since this issue was opened.

Am not advocating re-opening this (cause it seems kind of low priority atm), but am also not sure that it's fixed.

eenblam, 2018-06-13:
@bennlich yeah, you're right. I was in a hurry the other day, sorry.

Also agree that it may not be worth reopening, at least not without confirming that it's still assigning IPs in our subnet.

yardenac, 2018-06-13:
Sudoroom's old router had some l2tp stuff set up, but it wasn't actually being used. When I set up our new router last year, I didn't copy over any of it.

Send me an ssh key if you'd like access to the new router :)

eenblam, 2018-06-13:
Thanks @yardenac ! Will let you know if/when this makes it into our queue.