sudomesh/sudowrt-firmware #85: Colocation for tunnel server

max-b, 2015-12-05:
We're definitely seeing traffic growth on our tunnel/exit server:

http://monitor.sudomesh.org/cacti/graph.php?action=view&local_graph_id=217&rra_id=all

[image]

Our digital ocean droplet will allow us 2TB per month, though we can scale that up by paying more:
https://www.digitalocean.com/pricing/

Ideally, though we'd finally find a reasonably priced colocation facility. Especially for before we do our first big public release.

max-b, 2016-01-11:
We've since received sponsorship from digital ocean and it seems as though we're ok on the DMCA front. I'm pushing this back considering that we'll probably just want to put an exit server next to whatever ip transit link we eventually get.

paidforby, 2018-10-10:
we have a new exit server at HE, and do not use digital ocean for any of our exit nodes, issue no longer relevant

closed 2018-10-10 by paidforby

bennlich, 2018-10-22:
Seems like the Richmond tower would be a great place for this.