sudomesh/sudowrt-firmware #91: Unable to establish PPPTP VPN Connection on peoplesopen.net network

thebalaa, 2016-01-22:
Device: iPhone 6s, MacBook
OS: iOS 9.2.1, OS X El Capitan

Using built in VPN Client to connect to PPPTP VPN Server does not work, see attached screenshot. 
Works as expected on private network. I suspect it could be related to running PPPTP over tunnel digger

[image]

max-b, 2016-01-25:
This is because pptp doesn't work natively behind a masquerading firewall (NAT). All traffic passes out of our exit server to the internet, where it's NATed, so pptp won't work naively. 

It looks like there's some way to enable "pptp passthrough" in the firewall rules, so I can look into that now.

max-b, 2016-01-25:
@thebalaa - I added some iptables conntrack modules which I think should fix the issue. I was just successfully able to connect to a VPN over pptp over here. Can you try it on your end?

thebalaa, 2016-01-26:
It works! Thanks, Max.

max-b, 2016-01-26:
awesome. I added this as an issue to the provisioning scripts:
https://github.com/sudomesh/mesh-playbooks/issues/1

closed 2016-01-26 by max-b