sudomesh/sudowrt-firmware #94: Don't forward traffic to upstream lan

max-b, 2016-03-09:
If someone puts a mesh node behind their lan, the current status is that it will allow forwarding from the mesh subnet to devices on their lan.

The simple and reasonably quick fix is to drop any traffic from the mesh subnet to any non-publicly routable subnet (except for the mesh subnet of course):

```
    # Don't forward any traffic from mesh to upstream lan                                                      
    iptables -A FORWARD -s $MESHNET -d $UPSTREAMLAN -j DROP  
```

Where `$UPSTREAMLAN` would be  `192.168.0.0/16` and `10.0.0.0/8` (we're already using 172.16.0.0/12 for our "private" network)

Juul, 2016-03-09:
On Tue, Mar 8, 2016 at 4:25 PM, maxb <notifications@github.com> wrote:

> If someone puts a mesh node behind their lan, the current status is that
> it will allow forwarding from the mesh subnet to devices on their lan.
>
> The simple and reasonably quick fix is to drop any [...]

Oh dang I thought I had a rule for that! Good catch. That must be the
problem Ed reported long ago.

-- 
marc/juul

max-b, 2016-03-11:
Should be fixed as of: https://github.com/sudomesh/makenode/commit/5489034d815997eabf32fe668570173dc2b3e32e

closed 2016-03-11 by max-b