sudomesh/sudowrt-firmware #96: Client isolation shouldn't be happening / iptables rules need to be re-written

Juul, 2016-06-20:
Must be a problem with the firewall script.

The forwarding rules in that script are also suspect. All of these rules are too broad and override later DROP rules:

```
    iptables -A FORWARD -i $MESH2 -s $MESHNET ! -d $PRIVNET -j ACCEPT
    iptables -A FORWARD -i $MESH5 -s $MESHNET ! -d $PRIVNET -j ACCEPT
    iptables -A FORWARD -i $TUN -s $MESHNET ! -d $PRIVNET -j ACCEPT
    iptables -A FORWARD -i $OPEN -s $MESHNET ! -d $PRIVNET -j ACCEPT
    iptables -A FORWARD -i $EXT1 -s $MESHNET ! -d $PRIVNET -j ACCEPT
    iptables -A FORWARD -i $EXT2 -s $MESHNET ! -d $PRIVNET -j ACCEPT
    iptables -A FORWARD -i $EXT3 -s $MESHNET ! -d $PRIVNET -j ACCEPT
    iptables -A FORWARD -i $EXT4 -s $MESHNET ! -d $PRIVNET -j ACCEPT
```

We should rewrite the forwarding section so it only has ACCEPT statements and make those as specific as possible.

The INPUT rules are also all fucked. They have all this stuff that's related to forwarding. The INPUT rules should only deal with traffic trying to access the router itself.

INPUT (more or less?) allow everything. FORWARD is the important one.

jhpoelen, 2018-03-07:
is this related to https://github.com/sudomesh/bugs/issues/1 ?

paidforby, 2018-10-10:
I think this was solved by https://github.com/sudomesh/sudowrt-firmware/commit/0b8961d82ed8e721c40cb58c2c6f02d6eff6dcb9 
Though this issue does have some relevant info regarding the meshrouting script, I'm going with merge with a new, more general issue #151

closed 2018-10-10 by paidforby