sudomesh/sudowrt-firmware #98: Prevent logging of sensitive info

Juul, 2016-08-02:
The nodes are logging which clients associate and disassociate + IP leases offered.

We want to enable this for debugging purposes but it should be default off. Perhaps completely disabling logging is a good idea or perhaps we can filter sensitive information

See:

https://wiki.openwrt.org/doc/uci/system

sierkje, 2017-05-31:
@Juul As I mentioned after last meeting -- but not exactly what I remembered, and perhaps not as relevant: [https://github.com/EFForg/cryptolog](https://github.com/EFForg/cryptolog)

jnny, 2018-06-18:
status update?

paidforby, 2018-10-10:
Logging is still enabled, https://github.com/sudomesh/sudowrt-firmware/blob/master/files/etc/config/system, I suggest commenting it out and then including a script that can be run on the node to uncomment the relevant lines.

paidforby, 2018-10-10:
example of how to do this in shell using sed, https://github.com/sudomesh/sudowrt-firmware/blob/master/files/opt/mesh/pw_reset#L37