sudomesh/sudowrt-firmware #99: led stops working after running makenode

miloh, 2016-08-03:
updated firmware from builds.sudomesh.org on aug 02 2016, after running makenode (c648e084) on my debian jesse system, the router leds stopped lighting up. They look like they work because they light up after rebooting.

bennlich, 2018-02-26:
After some poking around w/ @paidforby, I think this might because of LED config missing from [/etc/config/system](https://github.com/sudomesh/sudowrt-firmware/blob/master/files/etc/config/system). I'll look at the config of a factory default router and see if we're missing some sensical defaults.

paidforby, 2018-04-21:
Zeroconf branch addresses this issue. Otherwise this is a problem with makenode, not sudowrt. If y'all would still like support for leds after running makenode, please open an issue in the makenode repo, or manually add the led configs shown in this file [zeroconf:/files/etc/config/system](https://github.com/sudomesh/sudowrt-firmware/blob/823bc5118e84e9968e0004d4347f5dce760cb094/files/etc/config/system) to `/etc/config/system` on your node.

closed 2018-04-21 by paidforby

bennlich, 2018-04-23:
Woot! Awesome.

commit 823bc511 (2018-04-21): adding led support for n600 fixes #99