sudomesh/sudowrt-firmware #101: Unable to connect to Apple services

redconfetti, 2016-08-12:
I'm using my private SSID on my router and for some reason I am unable to access any Apple services because of some issue with HTTPS communications. I'm unable to install apps using my iPhone, or access any Apple run website.

Doesn't happen when using 4G/LTE, or with the WIFI connection at work.

[image]

[image]

Juul, 2016-08-17:
Thanks a lot for the bug report! This is definitely related to the fake captive portal. Wanna help locate the problem? I'll be in for sudo mesh at 7:30 pm tonight :)

redconfetti, 2016-08-17:
It will have to be next Tuesday or Thursday.