sudomesh/sudowrt-firmware #104: Not able to build using new docker work flow?

gcgallo, 2017-04-08:
I'm having trouble building the firmware with whatever recent changes were made to the docker workflow. 

On arch, it doesn't even start the build (but that might be an arch problem)

On ubuntu, I get dependency errors for git-core and libssl-dev (will add console output, once I'm back on that machine)

On debian, I get a bunch on errors like this:
```
E: Failed to fetch http://archive.ubuntu.com/ubuntu/pool/main/libx/libxml2/libxml2_2.9.1+dfsg1-3ubuntu4.8_amd64.deb  404  Not Found [IP: 91.189.88.162 80]

E: Failed to fetch http://archive.ubuntu.com/ubuntu/pool/main/l/linux/linux-libc-dev_3.13.0-107.154_amd64.deb  404  Not Found [IP: 91.189.88.162 80]
```
On that very same debian machine, I was able to successfully build the firmware back in January. I tried again today using that same older repo (it was still on the machine, unaltered) and it's working just fine.

Any ideas?

andrewdollard, 2017-04-08:
Don't know about most of these, but I had some trouble with the git and
libssl dependencies the other day when building in the docker container. I
just updated the dockerfile to remove the version specifications for those
two and it worked.

gcgallo, 2017-04-08:
Thanks, that solution worked for ubuntu. Just got rid of the version specification for git-core and libssl-dev. Will push that change to the repo, unless there's any reason not to. I'm still building, so I'll wait to see how it turns out.

andrewdollard, 2017-04-08:
Cool. Feel free to push any changes. I’m actively working on this right now too, and have a couple other changes in the pipeline. Right now I’m trying to get the firmware to build in circleCI so we can just push to master and have fresh firmware built for us automatically :)

gcgallo, 2017-04-09:
Worked great, pushed updated Dockerfile. CI would be cool and useful...and easier than docker? maybe?

closed 2017-04-09 by gcgallo