sudomesh/sudowrt-firmware #107: Traffic prioritization

jtremback, 2017-09-20:
It came up at a meeting that someone would like to have prioritization so that their private traffic takes priority over mesh traffic. This would also allow the mesh traffic to take 100% of the capacity when they are not using the node. 

Here's some of what I know about shaping.

The linux tool `tc` does shaping. It has a bunch of different shaping tools called "qdiscs". The qdiscs nest inside each other, which is useful but kind of hard to understand.

I've used the DRR disc, which lets you prioritize different types of traffic at different rates. However, [I've found](http://altheamesh.com/blog/dev-update-2/) that DRR doesn't work without setting a  parent qdisc with an absolute maximum rate. This makes configuration more involved.

[PRIO](https://linux.die.net/man/8/tc-prio) might be another option. It looks more straightforward because it will set one type of traffic as being of a higher absolute priority than others. This is ideal for our use case and it looks like it is easier to set up than DRR as well.

jhpoelen, 2017-09-20:
related to https://github.com/sudomesh/bugs/issues/2 .

muxlux, 2017-09-20:
> PRIO <https://linux.die.net/man/8/tc-prio> might be another option. It
> looks more straightforward because it will set one type of traffic as being
> of a higher absolute priority than others. This is ideal for our use case
> and it looks like it is easier to set up than DRR as well.
>
> +1
~r