sudomesh/sudowrt-firmware #108: opkg feeds are wrong

Juul, 2017-10-10:
`/etc/opkg.conf` is wrong (referencing barrier breaker instead of chaos calmer). This is because it is wrong in makenode.

`/etc/opkg/distfeeds.conf` is wrong. The `chaos_calmer_peopleswifi` and `chaos_calmer_wlanslovenija` URLs are wrong. Not sure how this is generated. For a temporary fix, commenting out everything in distfeeds.conf and replacing the content of opkg.conf with the following should work:

```
src/gz chaos_calmer_base http://downloads.openwrt.org/chaos_calmer/15.05.1/ar71xx/generic/packages/base
src/gz chaos_calmer_luci http://downloads.openwrt.org/chaos_calmer/15.05.1/ar71xx/generic/packages/luci
src/gz chaos_calmer_packages http://downloads.openwrt.org/chaos_calmer/15.05.1/ar71xx/generic/packages/packages
src/gz chaos_calmer_routing http://downloads.openwrt.org/chaos_calmer/15.05.1/ar71xx/generic/packages/routing
src/gz chaos_calmer_telephony http://downloads.openwrt.org/chaos_calmer/15.05.1/ar71xx/generic/packages/telephony
src/gz chaos_calmer_management http://downloads.openwrt.org/chaos_calmer/15.05.1/ar71xx/generic/packages/management
src/gz chaos_calmer_wlanslovenija https://builds.sudomesh.org/builds/chaos_calmer/ar71xx/packages/wlanslovenija
src/gz chaos_calmer_peopleswifi https://builds.sudomesh.org/builds/chaos_calmer/ar71xx/packages/peopleswifi
dest root /
dest ram /tmp
lists_dir ext /var/opkg-lists
option overlay_root /overlay
```

jerkey, 2017-10-10:
# opkg update
[successful download of first six URLs]
Downloading https://builds.sudomesh.org/builds/chaos_calmer/ar71xx/packages/wlanslovenija/Packages.gz.
wget: can't execute 'openssl': No such file or directory
wget: error getting response: Connection reset by peer
Downloading https://builds.sudomesh.org/builds/chaos_calmer/ar71xx/packages/peopleswifi/Packages.gz.
wget: can't execute 'openssl': No such file or directory
wget: error getting response: Connection reset by peer
Collected errors:
 * opkg_download: Failed to download https://builds.sudomesh.org/builds/chaos_calmer/ar71xx/packages/wlanslovenija/Packages.gz, wget returned 1.
 * opkg_download: Failed to download https://builds.sudomesh.org/builds/chaos_calmer/ar71xx/packages/peopleswifi/Packages.gz, wget returned 1.

jerkey, 2017-10-10:
```
18:12 < juul> oh :( yeah we should allow http o giess
18:13 < juul> you can just comment out the two builds.sudomesh.org lines for now
18:13 < juul> i'll fix
18:14 < juul> ok i fixed it. now it no longer forces https
18:19 < jerkey> god it's slow, but i guess it worked, thanks!

jerkey, 2017-10-16:
`so doing the above works and "opkg update" works, but i still can't "opkg install kmod-usb-acm"

```
# vim customfeeds.conf 
# vim distfeeds.conf 
# cd ..
# vim opkg.conf
# opkg update
Downloading http://downloads.openwrt.org/chaos_calmer/15.05.1/ar71xx/generic/packages/base/Packages.gz.
Updated list of available packages in /var/opkg-lists/chaos_calmer_base.
Downloading http://downloads.openwrt.org/chaos_calmer/15.05.1/ar71xx/generic/packages/luci/Packages.gz.
Updated list of available packages in /var/opkg-lists/chaos_calmer_luci.
Downloading http://downloads.openwrt.org/chaos_calmer/15.05.1/ar71xx/generic/packages/packages/Packages.gz.
Updated list of available packages in /var/opkg-lists/chaos_calmer_packages.
Downloading http://downloads.openwrt.org/chaos_calmer/15.05.1/ar71xx/generic/packages/routing/Packages.gz.
Updated list of available packages in /var/opkg-lists/chaos_calmer_routing.
Downloading http://downloads.openwrt.org/chaos_calmer/15.05.1/ar71xx/generic/packages/telephony/Packages.gz.
Updated list of available packages in /var/opkg-lists/chaos_calmer_telephony.
Downloading http://downloads.openwrt.org/chaos_calmer/15.05.1/ar71xx/generic/packages/management/Packages.gz.
Updated list of available packages in /var/opkg-lists/chaos_calmer_management.
Downloading http://builds.sudomesh.org/builds/chaos_calmer/ar71xx/packages/wlanslovenija/Packages.gz.
Updated list of available packages in /var/opkg-lists/chaos_calmer_wlanslovenija.
Downloading http://builds.sudomesh.org/builds/chaos_calmer/ar71xx/packages/peopleswifi/Packages.gz.
Updated list of available packages in /var/opkg-lists/chaos_calmer_peopleswifi.
# opkg install kmod-usb-acm
Installing kmod-usb-acm (3.18.23-1) to root...
Downloading http://downloads.openwrt.org/chaos_calmer/15.05.1/ar71xx/generic/packages/base/kmod-usb-acm_3.18.23-1_ar71xx.ipk.
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-usb-acm:
 * 	kernel (= 3.18.23-1-b2f200610f46d20ef52d269421369d0c) * 
 * opkg_install_cmd: Cannot install package kmod-usb-acm.
# uname -a
Linux jerkey-tl-wdr3500-2 3.18.23 #1 Fri Dec 25 00:14:49 EST 2015 mips GNU/Linux

papazoga, 2017-10-16:
The likely culprit for opkg.conf is in makenode.