sudomesh/sudowrt-firmware #110: Build fails using docker on Ubuntu 14.04

Juul, 2017-10-17:
@paidforby any idea how to fix this?

```
Reading package lists...
 ---> d2f300e91c40
Removing intermediate container 8948da29e358
Step 3/8 : RUN DEBIAN_FRONTEND=noninteractive apt-get install -yq       build-essential=11.6ubuntu6       subversion=1.8.8-1ubuntu3.2       libncurses5-dev=5.9+20140118-1ubuntu1       zlib1g-dev=1:1.2.8.dfsg-1ubuntu1       gawk=1:4.0.1+dfsg-2.1ubuntu2       gcc-multilib=4:4.8.2-1ubuntu6       flex=2.5.35-10.1ubuntu2       git-core       gettext=0.18.3.1-1ubuntu3       quilt=0.61-1       ccache=3.1.9-1       libssl-dev       xsltproc=1.1.28-2build1       unzip=6.0-9ubuntu1.5       python=2.7.5-5ubuntu3       wget=1.15-1ubuntu1.14.04.2
 ---> Running in 50412da455a3
Reading package lists...
Building dependency tree...
Reading state information...
E: Version '1.8.8-1ubuntu3.2' for 'subversion' was not found
The command '/bin/sh -c DEBIAN_FRONTEND=noninteractive apt-get install -yq       build-essential=11.6ubuntu6       subversion=1.8.8-1ubuntu3.2       libncurses5-dev=5.9+20140118-1ubuntu1       zlib1g-dev=1:1.2.8.dfsg-1ubuntu1       gawk=1:4.0.1+dfsg-2.1ubuntu2       gcc-multilib=4:4.8.2-1ubuntu6       flex=2.5.35-10.1ubuntu2       git-core       gettext=0.18.3.1-1ubuntu3       quilt=0.61-1       ccache=3.1.9-1       libssl-dev       xsltproc=1.1.28-2build1       unzip=6.0-9ubuntu1.5       python=2.7.5-5ubuntu3       wget=1.15-1ubuntu1.14.04.2' returned a non-zero code: 100
```

paidforby, 2017-10-18:
Issue is most likely with the [Dockerfile](https://github.com/sudomesh/sudowrt-firmware/blob/master/Dockerfile). Try removing the version specification for the subversion package. I had a similar problem in #104.

paidforby, 2017-10-23:
@Juul if you have not solved this yet, feel free to push the necessary changes to the repo (unless the only changes needed were in [this commit] (https://github.com/sudomesh/sudowrt-packages/commit/7b4d116dd09617a1adf6fb1246c03cf296f3dadc) to sudowrt-packages), and I can try building the firmware overnight since I've had success with my current machine.

paidforby, 2017-10-25:
I did not have this problem with my machine. Though I am now seeing this error during the build, when it is pulling in the feeds, 
```
fatal: No tags can describe '5f0540c370d5ab443d7cc3de523b19d1e49b4097'.
Try --always, or create some tags.
```
not sure if it will cause errors in the build, waiting for it to finish.

jhpoelen, 2017-11-05:
I was able to reproduce 
```E: Version '1.8.8-1ubuntu3.2' for 'subversion' was not found```

In addition, I found error:

```
E: Version '1.15-1ubuntu1.14.04.2' for 'wget' was not found
```

jhpoelen, 2017-11-05:
upgrading subversion and wget packages seems to make the errors go away . . . and was able to successfully build docker container for building the firmware. 

Here's the result on my command-line after executing ```sudo docker build -t sudomesh/sudowrt .
``` - 

```
Removing intermediate container f62883570551
Step 8 : ENTRYPOINT ./entrypoint.sh
 ---> Running in 21b85891e51f
 ---> b8c26c3607bd
Removing intermediate container 21b85891e51f
Successfully built b8c26c3607bd
```

jhpoelen, 2017-11-05:
@paidforby I was able to reproduce your errors when running ```docker run -v $PWD/firmware_images:/firmware_images sudomesh/sudowrt ar71xx``` , snippet from terminal output -

```
[...]
Importing feeds
+ cd /usr/local/sudowrt-firmware/built_firmware/openwrt
+ echo 'Importing feeds'
+ ./scripts/feeds update -a
Updating feed 'peopleswifi' from 'git://github.com/sudomesh/sudowrt-packages.git' ...
Cloning into './feeds/peopleswifi'...
Create index file './feeds/peopleswifi.index' 
fatal: No tags can describe 'bbed69b2dbb8b3afe93fa6d349b21717e5856e76'.
Try --always, or create some tags.
[...]
```

jhpoelen, 2017-11-05:
Looks like the scary looking error ```fatal: No tags can describe 'bbed69b2dbb8b3afe93fa6d349b21717e5856e76' ``` related to a commit on chaos_calmer branch of openwrt that does not have a tag associated where the feed update scripts expects it. See https://github.com/openwrt/openwrt/commit/bbed69b2dbb8b3afe93fa6d349b21717e5856e76 for commit.

jhpoelen, 2017-11-05:
previously, openwrt version used was 15.05:LATEST . By setting the version to 15.05:0c335d6ea00af9b32225f72d491e560c1a7cf4ab the most recent commit with a tag is used. However, I decided to revert the "fix" because the last commit on 15.05 branch of openwrt with a release tag is dated 28 Jan 2016 and assumed that builds have been happening ok after that.

jhpoelen, 2017-11-06:
After about 10 hours, I was able to build an (untested) version of the ar71xx firmware. Hoping to reproduce in a build environment #111 at some point.  

@Juul @paidforby can you verify?

exit code from docker image:
```
9a86a8781f6f        325b93529e11        "./entrypoint.sh ar71"   15 hours ago        Exited (0) 10 hours ago                         trusting_allen
```

jhpoelen, 2017-11-06:
Please note that I failed to find any of the n600 firmware packages. Attaching list of produced firmware. Further investigation needed to figure out why n600 packages didn't show up.

[ar71xx-firmware-list.txt](https://github.com/sudomesh/sudowrt-firmware/files/1446659/ar71xx-firmware-list.txt)

paidforby, 2017-11-06:
@jhpoelen thanks for looking into to this. I will try to run a test build this afternoon. For testing, I'd recommend removing the last two lines of the [entrypoint.sh](https://github.com/sudomesh/sudowrt-firmware/blob/master/entrypoint.sh) (be careful of commenting, as the docker build can treat them oddly). That way, you'll only build the home node firmware which should only take a few hours (not 10!). I've played around with the entrypoint.sh to get something more general, but left it hardcoded because I'd didn't see a use case where we wouldn't want to build both the home-node and extender-node firmware (except testing), see the [README](https://github.com/sudomesh/sudowrt-firmware#the-easy-way) and #105 for more on that.

jhpoelen, 2017-11-08:
original issue related to subversion / wget dependencies was fixed. Closing issue.

closed 2017-11-08 by jhpoelen

paidforby, 2017-11-08:
I can confirm @jhpoelen results. However, per my own suggestion, I removed the extender node build. After ~three hours, the build appeared to succeed, only to provide a set of firmware images that does not contain the n600. It looks like it is only building some of the images, (perhaps tied to that fatal error?). Here's the snippet from the end of my build.

```
Building
+ openwrt_builder ar71xx
+ local arch=ar71xx
Building ar71xx in dir: /usr/local/sudowrt-firmware/built_firmware/builder.ar71xx
+ local build_dir=/usr/local/sudowrt-firmware/built_firmware/builder.ar71xx
+ echo 'Building ar71xx in dir: /usr/local/sudowrt-firmware/built_firmware/builder.ar71xx'
+ make V=s -C /usr/local/sudowrt-firmware/built_firmware/builder.ar71xx

real	119m3.745s
user	70m11.308s
sys	9m31.012s
```

However, despite my changes to entrypoint.sh, it still attempted to build the extender node firmware, but it failed.
```
+ echo 'Building ar71xx for extender node in dir: /usr/local/sudowrt-firmware/built_firmware/builder.ar71xx.extender-node'
+ make V=s -C /usr/local/sudowrt-firmware/built_firmware/builder.ar71xx.extender-node
Building ar71xx for extender node in dir: /usr/local/sudowrt-firmware/built_firmware/builder.ar71xx.extender-node
Image build for extender node for architecture 'ar71xx' has failed. See build.extender-node.log for details!
+ '[' 2 '!=' 0 ']'
+ echo 'Image build for extender node for architecture '\''ar71xx'\'' has failed. See build.extender-node.log for details!'
+ exit 1

real	35m38.338s
user	26m43.772s
sys	4m33.604s
```