sudomesh/sudowrt-firmware #118: create operator manual

jhpoelen, 2017-12-12:
Create operator manual to help folks like myself troubleshoot home, extender and exit nodes. 

Content would include: network diagram (how is a home node configured), troubleshooting recipes (how to check that a node is meshing?).

Please holler if such a manual already exists.

paidforby, 2017-12-12:
I agree, an definitive operator's manual would be nice. I am interested on working on the needed documentation and test systems. I would also like to understand how to put something on the mesh once I have a node? Seems like a logical next step, hence my reason for opening [bug 9](https://github.com/sudomesh/bugs/issues/9).

jhpoelen, 2017-12-13:
@paidforby here's a first pass https://github.com/sudomesh/sudowrt-firmware/blob/master/operator_manual.md . I can imagine that setting up a hello world service could be part of this.

closed 2018-03-07 by jhpoelen