sudomesh/sudowrt-firmware #119: auto clean-up docker container on successful build

paidforby, 2017-12-15:
Can we add an option to automatically clean up the docker container after successfully building the firmware? Should this go at the end of the entrypoint.sh or must it be run outside the container? Ideally, it would run only on a "successful" build and would only delete or modify the container and image just used to build that image.

paidforby, 2018-03-07:
Broadening scope of this issue:
After running sudowrt firmware build "the easy way" three or four times, my builds machine runs out of disk space. This is also related to #111 since I suspect this with be an issue on a dedicated builds machine. I will spin up a clean builds server to see if I see the same behavior there.

gobengo, 2018-03-07:
Would adding `--rm` to the `docker run ...` command in the README help with what you're talking about?

`docker run -v $PWD/firmware_images:/firmware_images sudomesh/sudowrt-firmware:0.2.1 --rm`

Usually you don't want to do this in the entrypoint, because otherwise you have erased information that might be useful to go explore when debugging failed (or even successful) `docker run`s

paidforby, 2018-03-07:
Maybe? I will try another build later and see it if helps, thanks for the tip

paidforby, 2018-03-10:
Thanks for that tip @gobengo That appears to solve my problem, however, the build fails on the cheapest digital ocean droplet (25GB storage) because it runs out of space (it also takes insanely long). Any ideas? Should I just build the "hard way" since I'm spinning up a DO droplet anyway? Maybe I could create the droplet from our docker hub image?

gobengo, 2018-03-11:
I'm pretty sure that in here were building the kernel from source and
plenty of other packages too. I'm sitting here in an Embedded Linux course
and the system recommendations for doing the same were 4 cores, 16Gb
memory, and 100Gb storage, and all on bare metal.

Running outside of docker may help a bit, but not that much. Overhead of
Linux containers shouldn't be that much (compared to the virt layer that
you have no choice about because cloud).
Try the same way on a more expensive droplet.

paidforby, 2018-08-06:
Since learning more about the ways of docker through reworking the build process in #137, I'm considering this issue closed. I probably just need to be more careful about destroying containers after use and deleting images after pulling/building. Thanks for the help @gobengo.

closed 2018-08-06 by paidforby