sudomesh/sudowrt-firmware #120: Tag a release for new build of firmware

paidforby, 2017-12-15:
Start tagging stable commits and putting them on [sudowrt/releases](https://github.com/sudomesh/sudowrt-firmware/releases), Update version number to v0.3? or at least v0.2.1? Match in https://github.com/sudomesh/sudowrt-firmware/blob/master/files/etc/banner, suggested name for next version is "left hand of darkness".

jhpoelen, 2017-12-15:
Assuming that we haven't added any major features, I'd go for v0.2.1. Did you check whether the tags propagate to docker hub build images?

paidforby, 2017-12-16:
should have probably been updated to v3 when the new dashboard was added. Anyway, we should keep better track of stable releases and keep updating the banner so people know what version they are running. I will revert the banner and make it v0.2.1. I'm not sure about your docker hub question, can we/should we make it pull from a stable commit rather then from HEAD? Or make it build both?

paidforby, 2017-12-16:
Latest commit tagged v0.2.1. Tags do appear to propagate to TravisCI, see https://travis-ci.org/sudomesh/sudowrt-firmware/builds; however, it does not appear to automatically propagate to docker hub. I believe this can be set in dockerhub through its gui, see https://stackoverflow.com/questions/25328166/docker-hub-automated-build-tagging, I do not believe I have access to the docker hub to make the necessary changes.

jhpoelen, 2017-12-16:
If you share docker hub user name via signal I can add you.

jhpoelen, 2017-12-16:
I've enabled tag releases via [.travis.yml](../blob/master/.travis.yml). And, have created sudomesh docker org to enable multi-user collaboration. So, ```docker pull sudomesh/sudowrt-firmware``` instead of ```docker pull sudowrt/firmware```.  Also, I've created a release https://github.com/sudomesh/sudowrt-firmware/releases/0.2.1 (deleted/created cycle after moving to sudomesh docker hub org).

paidforby, 2017-12-16:
Thanks!  Looks great, https://hub.docker.com/r/sudomesh/sudowrt-firmware/tags/ 
My docker id is "paidforby".
Is this release confirmed to be working? I started a build earlier today and it failed as documented in #116. I thought that error only occurred when using ```--net=none``` switch on docker run?

jhpoelen, 2017-12-16:
Ok, just added you to sudomesh org. And yes, that error should only occur on not having a network connection. Please add logging to #116 if this is not the case.

paidforby, 2017-12-16:
Confirmed, building with 0.2.1 per README, succeeds, Thanks @jhpoelen

closed 2017-12-16 by paidforby