sudomesh/sudowrt-firmware #121: Ports on mynet n600 not providing IP addresses

paidforby, 2017-12-16:
Only port 3 on mynet n600 flashed with sudowrt v0.2.1 appears to be working. Port 1, 2, and 4 do not appear to provide any IP address or network access.  According to the [operator's manual](https://github.com/sudomesh/sudowrt-firmware/blob/master/operator_manual.md#mesh-network-components) port 1 and 2 are reserved for meshing (through extender nodes?) so I can see why those may not provide IPs. But shouldn't port 4 provide a "100." mesh ip address?

jhpoelen, 2017-12-16:
Can you get an internet connection through the peoplesopen ssid on the router?

paidforby, 2017-12-16:
Have not checked connection to exit node since reflashing, will try later and report back

paidforby, 2017-12-17:
Confirmed that I am able to access the internet via the exit node on the peoplesopen ssid and am receiving a mesh ip address over wifi. Still nothing over ethernet port 4.

I considered trying to set a static ip as suggested by @juul in [bug 9](https://github.com/sudomesh/bugs/issues/9). I believe the dnsmasq config referenced  is located in `/etc/init.d/dnsmasq` on a newly flashed node.
I found this in said file at line 409:
```
# Do not support non-static interfaces for now
[ static = "$proto" ] || return 0       
```
I am unable to find this exact config any of our repos? It seems that dnsmasq is pulled in by makenode [here](https://github.com/sudomesh/makenode/blob/c648e08462b16aeb6532559b3b49067d2b2d4745/configs/templates/files/etc/config/dhcp) and [here](https://github.com/sudomesh/makenode/blob/c648e08462b16aeb6532559b3b49067d2b2d4745/configs/templates/files/etc/resolv.conf.dnsmasq). Does this spark any ideas?

paidforby, 2017-12-17:
Setting an arbitrary static ip does not seem to work, i can't seem to find the reserved block of ips mentioned in [bug 9](https://github.com/sudomesh/bugs/issues/9) there is also [`/etc/dnsmasq.conf`](https://github.com/sudomesh/makenode/blob/c648e08462b16aeb6532559b3b49067d2b2d4745/configs/ar71xx/home_nodes/mynet_n/templates/files/etc/dnsmasq.conf) also configured by makenode.

jhpoelen, 2017-12-17:
To put service on mesh, you'd have to use a mesh ip right? Please confirm that operator's manual description of port 3 and port 4 should be swapped.

Juul, 2017-12-17:
The ports on some mynets are scrambled, meaning that it's not port 1 and 2
that are extender node ports. Extender node ports will not give you an IP.
One port should give you a 100.64 IP and the other should give you a 172.30
IP. The remaining two will not give you an IP unless you are an extender
node.

paidforby, 2017-12-17:
I received a 172.30 IP (private) from port 3. I did not receive an IP from any of the other ports. Can someone confirm what I am seeing? I will try flashing another node, perhaps something odd about mine.

@jhpoelen, description of port 3 in operator's manual is correct, though, I cannot confirm it is correct for the other ports.

paidforby, 2018-10-10:
Ports work as described by Juul above. We even have a sticker guide thanks to Santa Cruz meshnet people https://github.com/sudomesh/propaganda/blob/master/stickers/router_ports/santa_cruz_meshnet_mynet_n600_labels_for_avery_5264.svg

closed 2018-10-10 by paidforby