sudomesh/sudowrt-firmware #122: Can't connect to 172.22.0.1 on WDR4300

alexgleason, 2017-12-19:
Hi all, I've flashed sudowrt on a WDR4300 following Mesh/WalkThrough.

The router's WAN port is connected to a LAN port on an internet-enabled router. The router's 1st LAN port is connected to my Ubuntu 16.04 laptop. With DHCP enabled on the laptop I could not connect to 172.22.0.1 to complete the installation. So, I tried setting up a static IP address:

enp0s25
inet addr: 172.22.0.10
bcast: 0.0.0.0
mask: 255.255.255.0

^ output from ifconfig on my laptop. I've also disabled wifi on my laptop.

I still cannot connect to 172.22.0.1 from my browser (nor https://172.22.0.1). I can ping the address and receive packets, though.

The lights on the router come on, and the wired connection light turns on when the cable is plugged into my laptop. I think the firmware flashed correctly. I'm not sure why it will not connect, though.

Any ideas/direction/links? Thanks!

yardenac, 2017-12-19:
Same thing happened to me earlier this week - I found the interface at https://172.30.0.1 instead

alexgleason, 2017-12-19:
Thanks for the quick reply.

When I connect through DHCP the network manager GUI says my router is located at 172.22.0.1.

* When I try to visit 172.22.0.1, my browser says ERR_CONNECTION_REFUSED.
* When I visit 172.30.0.1 my browser says ERR_ADDRESS_UNREACHABLE.
* I can ping 172.22.0.1 no matter what, while I cannot ping 172.30.0.1.

This leads me to believe that 172.22.0.1 should be the right address.

I've also tried setting a static IP to 172.30.0.10 and I get the same results.

alexgleason, 2017-12-19:
Apparently "connection refused" probably means that [no process is listening on that port](https://serverfault.com/questions/725262/what-causes-the-connection-refused-message#725263).

I get "connection refused" when trying to connect via http, https, and ssh. This is making me think that maybe the install did go poorly and the router isn't running the necessary services (openssh, webserver, etc).

jhpoelen, 2017-12-19:
Interesting! I far as I know, the firmware has wireless disabled by default. 

Did you consider installing vanilla openwrt to see whether it is a compatibility issue? https://wiki.openwrt.org/toh/tp-link/TL-WDR4300 .

Also, I'd invite you to come over tonight for our meeting . . . but I am noticing your are on the east coast...