sudomesh/sudowrt-firmware #123: sudowrt firmware broken - 15.05 branch of openwrt no longer seems to exist

jhpoelen, 2018-01-04:
```git clone git://git.openwrt.org/15.05/openwrt.git```

worked up until about 2 days ago (~1 jan 2018, see attached screenshot).

[image]

paidforby, 2018-01-06:
Thanks for reporting, just noticed this also. Who deleted the 15.05 branch, we were using that!

paidforby, 2018-01-06:
attempting to fix with:
```
git clone "git://github.com/openwrt/chaos_calmer.git"
```

Now failing at:
```
oldpackages:src-git oldpackages git://git.openwrt.org/15.05/openwrt.git
```
in [openwrt_config/feeds](https://github.com/sudomesh/sudowrt-firmware/blob/master/openwrt_config/feeds) is also broken by this.

closed 2018-01-06 by paidforby

paidforby, 2018-01-06:
All references to `git://git.openwrt.org/15.05/openwrt.git` have been replaced with `git://github.com/openwrt/chaos_calmer.git` still building off tagged commit `0c335d6ea00af9b32225f72d491e560c1a7cf4ab`. Need to test full build to see if there were any adverse side effects, particularly with the oldpackages switch, but for now it is passing.

paidforby, 2018-01-06:
side note: looks like 15.05 was archived at `https://git.archive.openwrt.org/15.05/openwrt.git`

paidforby, 2018-01-06:
Apologies for all the commits and reverts, decided to make simplest change, inserted "archive" between "git" and "openwrt" as seen in 6c16f45bb80a4daef27bfa2058d482ca364bfcb5

jhpoelen, 2018-01-06:
thanks for making this happen. Does that mean that the head commit stays the same?

jhpoelen, 2018-01-06:
btw - should we turn travis notifications on for all contributors? Now, I am not catching the build that you triggered..

paidforby, 2018-01-06:
I'm not sure what is meant by them "archiving" it if that is what you are asking? It looks like the [openwrt/chaos_calmer](github.com/openwrt/chaos_calmer.git) repo is still being developed.

I don't mind getting notifications on failed builds, but I guess it should be an opt-in, right.

Also, we should move the 0.2.1 tag to 6c16f45bb80a4daef27bfa2058d482ca364bfcb5, since the old stable commit will now fail.

jhpoelen, 2018-01-06:
I'd suggest to create a new tag. In my mind , the tag is for the docker build image.

jhpoelen, 2018-01-06:
My question about the commit was whether the openwrt commit hash we use now has changed. I can also have a look at the version file in the build repo. Just being lazy.

paidforby, 2018-01-06:
understood on the tag, whoever gets to it first can tag it accordingly 

we're still checking out the same commit, for your convenience https://github.com/sudomesh/sudowrt-firmware/blob/master/openwrt_config/version, as far as I can tell there is no difference between the old, active openwrt 15.05 repo and the archived repo (that is, it still recognizes the same commits and tags).

commit 4726575d (2018-01-06): pull 15.05 from github, attempt to fix #123
commit 2cef9d27 (2018-01-06): updated oldpackages feed to github repo, actually fixes #123
commit 8313f3c8 (2018-01-06): Revert "updated oldpackages feed to github repo, actually fixes #123"

This reverts commit 2cef9d275932186a49802569c11b9d76b6505f95.
commit ab02fd37 (2018-01-06): Revert "pull 15.05 from github, attempt to fix #123"

This reverts commit 4726575dc684fb6a54a70865782e94e24f8484ba.