sudomesh/sudowrt-firmware #124: No way to log into extender nodes

Juul, 2018-02-07:
There is currently no way to log into an extender node via ssh. Since there is not configuration step that means that there is no password set. How can we make it possible to log in without a configuration step? Could we have an alternate extender node image with the only difference being that it has a known default password set? Then people who want ssh access can use that image instead?

paidforby, 2018-02-15:
I'm actively looking into this. Can we just roll `/etc/passwd` and `/etc/shadow` into the firmware build? This appears to be how makenode sets passwords for home nodes. We were discussing providing similar default functionality for home nodes anyway. I'll try with an alternate extended node build first, but will keep you updated as I work on documenting makenode.

Juul, 2018-02-15:
We should not be setting a default password for the extender nodes as this then requires people to log in and change it which is an extra step not unlike running `makenode` in difficulty and worse it's a security issue. We talked about having a default password on only the private port of the home node but I don't really see how we can do something like this for extender nodes since most have only one port. We could do it on first boot, or ((for the first 24 hours) or (until the extender node is plugged into a home node for the first time)) whichever happens first.

paidforby, 2018-02-15:
Certainly, this is precisely what I had in mind for both the home node and extender nodes. Both limit the default logon to a single port (obviously a problem for extender nodes) and to first boot or a specific time period (because what if you never turn off the node after flashing?). You suggested an alternative build of the firmware only to be used by people who want ssh access to their extender nodes, this is also a reasonable solution, but should be used/distributed with care. I'm running an test alt-build right now to see if my theories regarding `/etc/passwd` and `/etc/shadow' work out. Meanwhile, I'll look into that first boot/timeout condition.

paidforby, 2018-02-15:
7 hours later... can confirm that dropping ```passwd``` and ```shadow``` into ```/files/etc/```, placing an MD5 hashed password into ```shadow```, and then building the firmware enables that password as the "default" root password, allowing ssh access on home nodes without running makenode. Can cannot confirm for logging into the admin dashboard because I copied ```passwd``` from makenode and forget to replace the variable fields for the admin user. Have yet to test on extender node, but I imagine it should work the same. I'll run another test build to correct that admin mistake and to try using just crpyt3 for generating the MD5 hashed password, (the previous hash I copied from my personal home node, generated by makenode, using crypt3).

paidforby, 2018-02-15:
Inserting default password into `shadow` works with any MD5 hash (generated using crypt3?). However, the same method does not appear to work for logging into the admin dashboard, this maybe a problem with [peoplesopen-dash](https://github.com/sudomesh/peoplesopen-dash) itself, not sure if @jtremback has any ideas regarding this. Will test default root access on an extender node once I'm able to flash one with the alt-build, probably tonight.

paidforby, 2018-02-16:
Woo hoo! The alternate build worked on both homes nodes (n600) and extender nodes (nanobridge m5), the fork currently lives at https://github.com/paidforby/sudowrt-firmware. @Juul you can find a selection of the dev-builds (as I'm calling this alternate build) "hidden" on the builds server at https://builds.sudomesh.org/dev-builds/, I don't see why this shouldn't be public, as long as it comes with a warning. I'll continue working on the additional features we discussed and won't assume this issue closed until we address those.

paidforby, 2018-02-21:
Note: current dev-build breaks the ability to use makenode to configure a home node unless you change the root password in [makenode/settings.js](https://github.com/sudomesh/makenode/blob/master/settings.js.example) to match the dev-build default password.

paidforby, 2018-10-22:
Outdated, a lot of related progress was made during autoconf development. New attempts at extender node firmware should generate new issues.

closed 2018-10-22 by paidforby