sudomesh/sudowrt-firmware #129: zeroconf branch: tunneldigger will not start

paidforby, 2018-04-04:
On the latest build of the zeroconf firmware, available here http://builds.sudomesh.org/dev-builds/dispossessed/0.3.0/ , it appears as though tunneldigger is unable to find the tunnel broker and, therefore, cannot create the l2tp interface to provide internet access over the peoplesopen ssid. It seems as all the configurations have been correctly created, but maybe I missed something from makenode.

A good place to start would be doing a thorough, file-by-file comparison of a madenode and zeroconf'd node.

bennlich, 2018-04-06:
Hey! Trying to get set up with a zeroconf node, but I don't know the admin SSID or root ssh passwords. Are those written down anywhere?

paidforby, 2018-04-08:
Awesome, @bennlich  Most configurations are (or should be) stored in the [zeroconf script](https://github.com/sudomesh/sudowrt-firmware/blob/zeroconf/files/opt/mesh/zeroconf) for easy access. The root password is not there because it must be hashed before being put into [shadow](https://github.com/sudomesh/sudowrt-firmware/blob/zeroconf/files/etc/shadow) (this is something that makenode does, but I just hardcoded it, perhaps we could hash it during zeroconf). The hash should correspond to "praisebarb" if not try "meshtheplanet", we should treat the root pass for zeroconf as an "open secret" until we have a way of auto-disabling it after a certain period of time.

paidforby, 2018-04-17:
discovered a couple issues in template files, [/etc/config/tunneldigger](https://github.com/sudomesh/sudowrt-firmware/blob/zeroconf/files/opt/mesh/templates/etc/config/tunneldigger) had a typo, the interface should be `eth1` not `eth0` (at least for N600), additionally, the server ips need to be updated a la patch to bug 17. It also seems that zeroconf won't work with the HE exit node until we update the tunneldigger client in this branch. Finally, the dns options were screwed up in both [/etc/resolve.conf.dnsmasq](https://github.com/sudomesh/sudowrt-firmware/blob/zeroconf/files/opt/mesh/templates/etc/resolv.conf.dnsmasq) and [/etc/config/network](https://github.com/sudomesh/sudowrt-firmware/blob/zeroconf/files/opt/mesh/templates/etc/config/network), integrating the patch to bug 27 will solve this. Will apply necessary changes, rebuild the firmware, and test before closing this issue.

paidforby, 2018-04-17:
oh, I should also integrate the good parts of the patch to bug 23

paidforby, 2018-04-18:
Tested and working. Also confirmed that it is able to babel with other nodes and route to the internet via the exit node(s).

closed 2018-04-18 by paidforby