sudomesh/sudowrt-firmware #131: zeroconf branch: not able to access admin dashboard without running makenode

paidforby, 2018-04-18:
After zeroconfiguration is complete and the node restarted, you are still unable to access the admin dashboard, https://github.com/sudomesh/peoplesopen-dash, with the default admin password. This problem is also seen when attempting access the admin dash prior to running makenode (on the standard branch of the firmware). Not sure if the problem is with the firmware or the dashboard. The following question come to mind. How does makenode set the admin password? How does the dashboard authenticate this password?

paidforby, 2018-04-18:
nevermind, just checked, it is working just fine over the private network. However, it is not accessible over the public network. I believed this is the case in both standard and zeroconf. Can we/should we make the admin dash accessible over the public SSID?

paidforby, 2018-04-18:
I guess the main complaint is that is could get confusing if you type in the wrong IP address while on the public network. And of course, it may be an unnecessary security concern, especially if we add more features to the dash? I'll mark as an enhancement and leave it out of zeroconf for now, but would like to revisit it once we are working on improvements to the dashboard again.

paidforby, 2018-04-18:
also note for later, may also need to modify this file https://github.com/sudomesh/peoplesopen-dash/blob/master/src/config.js for the log in  to work over the public network

paidforby, 2018-04-18:
My theory is this: 
Why even have a private network? It's sole use is to access the admin dashboard. It only causes confusion to node owners who don't realize you can't mesh over it. For all intents and purposes, it is impossible to run WPA2 on ad-hoc interfaces, meaning no private meshing anytime soon, see https://github.com/sudomesh/bugs/issues/13.
Why would we even want private meshing? It runs counter to the values of the project and only gives node owners a false sense of security.
Finally, now with zeroconf extant, the private network one of few difference left between home nodes and extender node, see https://github.com/sudomesh/bugs/issues/29. So why not just forget about the difference, extender nodes are just as much routers and home nodes, let's zeroconf them also.

Note: these are my opinions, I'd love to hear counter arguments.

jhpoelen, 2018-04-18:
+1 for dropping private network .

bennlich, 2018-04-19:
Ahhh, interesting. Had not considered that option. So this would mean all node administration would occur through either a predetermined ethernet port, or through the public SSID. Is that right?

I guess you'd still have a WAN interface for digging tunnels, it just wouldn't be accessible from any of the node's wireless networks.

paidforby, 2018-04-19:
@bennlich yes, at first, we may want to make the admin dash accessible only through a specific ethernet port (I'd suggest port 3 on N600s). Then, later we can add access through the public network once we are more confident about the security and maybe have a more personalized log in screen (maybe a picture or name, so you can tell it's your node).

One valid argument I've heard in favor of keeping the private network is for IoT devices (e.g. Sir Alexas, and Madame Hue Bulbs) that you mostly likely don't want accessible via the mesh. My take on this, and private networks in general, is that peoplesopen nodes are intended to provide open access points that mesh physically or virtually, not personalized private networks for your home. If this is your use case and you don't already have a private home WiFi network, you can get any-old-off-the-shelf router and change the default network settings to set up a private WLAN. Additionally, if a new node owner needs help with this sort of set up during your peoplesopen node installation, they can ask a node whisperer or a PONI they'd (probably) be happy to help.

Oh, also if anyone needs "any-old-off-the-shelf router", we have a pile of Linksys WRT54Gs collecting dust.

I can also see dropping the private network as a motivating factor to make sure everyday use cases, like IoT stuff, Netflox, Hulus, etc., function properly over the public network.