sudomesh/sudowrt-firmware #133: pre-build fails to retrieve wget

paidforby, 2018-05-10:
The pre-build for the firmware is failing both on Travis-CI and locally, here's the error,
```
E: Version '1.15-1ubuntu1.14.04.3' for 'wget' was not found
```

paidforby, 2018-05-10:
removing version requirement on wget in [Dockerfile](https://github.com/sudomesh/sudowrt-firmware/blob/master/Dockerfile) resolves issue, but is probably not the best way of dealing with this, should figure out latest stable version of wget for ubuntu 14.04, same goes for git core and libssl-dev which we had previously band-aid fixed by removing version requirements.

paidforby, 2018-05-10:
Found this, here https://launchpad.net/ubuntu/+source/wget seems like a likely cause
```
Changelog

wget (1.15-1ubuntu1.14.04.4) trusty-security; urgency=medium

  * SECURITY UPDATE: Cookie injection vulnerability
    - debian/patches/CVE-2018-0494.patch: fix cooking injection
      in src/http.c.
    - CVE-2018-0494

 -- <email address hidden> (Leonidas S. Barbosa)  Tue, 08 May 2018 13:59:12 -0300
```

paidforby, 2018-05-11:
fixed wget, git-core, and libssl versions in both master and zeroconf

closed 2018-05-11 by paidforby

jhpoelen, 2018-05-13:
nice!