sudomesh/sudowrt-firmware #134: upgrade to babeld 1.8.1 (or later version)

paidforby, 2018-05-11:
Related to https://github.com/sudomesh/bugs/issues/31 and https://github.com/sudomesh/bugs/issues/33
We are in the process of deprecating our custom fork of babeld and begin using the main branch of babeld. To this we will need to,
* re-build firmware with main fork of babeld, see packages, https://github.com/sudomesh/sudowrt-firmware/blob/master/openwrt_config/packages#L45
* refactor any sudomesh/babeld dependent configs/scripts, potential problem files, https://github.com/sudomesh/sudowrt-firmware/blob/zeroconf/files/opt/notdhcpserver/hook.sh#L48 and https://github.com/sudomesh/sudowrt-firmware/blob/zeroconf/files/opt/mesh/templates/etc/init.d/meshrouting

jnny, 2018-06-18:
status update?

bennlich, 2019-05-05:
exitnodes today use babeld 1.8.2 (see https://github.com/sudomesh/exitnode/blob/5d142ec6be75d93ee239a0d3145e204799adf7f6/create_exitnode.sh#L33). It's backwards compatible with older versions of babeld.

closed 2019-05-05 by bennlich