sudomesh/sudowrt-firmware #138: can't log into dash on sudowrt 0.3.0

jnny, 2018-06-24:
Excited about this release and the debut of autoconf! (or is it zeroconf?)

Was able to successfully flash two MyNet N600s with the new firmware on Ubuntu 14.04, however am unable to login to the admin dashboard. Am able to ssh into the device as root using default password 'meshtheplanet', but same attempt on the admin panel fails.

bennlich, 2018-06-24:
Sanity check: are you accessing the dashboard over http or https? Last I checked, login only works over http.

bennlich, 2018-06-24:
I just flashed a node with 0.3.0 and was able to successfully log in to the admin dashboard over http with default creds.

jnny, 2018-06-25:
I was using http - as I had to add a security exception. Seems like the admin panel isn't loading properly in the browser...
@Juul can confirm.

bennlich, 2018-06-25:
Hmm. Could you send a screenshot of the entire browser window and errors in the JavaScript console?

jnny, 2018-06-26:
Looking for solutions, I discovered that the password for the private "pplsopen-admin" SSID is in fact "meshtheworld," not "meshtheplanet": https://github.com/sudomesh/sudowrt-firmware/blob/master/files/opt/mesh/autoconf [line 22]

Further down in that file, I noticed the IP "172.30.0.0" - and took that as a lead, after which I *was* actually able to log into a newly-flashed router by resetting the ethernet address to 172.30.0.10 and hitting http://172.30.0.1 in my browser. I've not been able to recreate this scenario, however, on a re-flashed N600 using Mint.

Screenshot and log of Mint attempting to connect via 172.22.0.1
(yes, all browser security disabled):

[image]

[debuglog_console_bug138.pdf](https://github.com/sudomesh/sudowrt-firmware/files/2135972/debuglog_console_bug138.pdf)
 
Will try resetting the ethernet IP again and using Chromium (tried on older Firefox and Quantum)

muxlux, 2018-06-26:
> files/opt/mesh/autoconf [line 22]

i was able to login to the panel at 172.30.0.1 too

-- 
*First they ignore you, then they laugh at you, then they fight you, then
you win*.
- Mohandas Karamchand Gandhi

CONFIDENTIALITY NOTICE:
The contents of this email message and any attachments are intended solely
for the addressee(s) and may contain confidential and/or privileged
information and may be legally protected from disclosure. If you are not
the intended recipient of this message or their agent, or if this message
has been addressed to you in error, please immediately alert the sender by
reply email and then delete this message and any attachments. If you are
not the intended recipient, you are hereby notified that any use,
dissemination, copying, or storage of this message or its attachments is
strictly prohibited.

bennlich, 2018-06-26:
@jnny in 0.3.0 you can log in w/o ethernet by connecting to the private SSID and pointing your browser to http://172.30.0.1

To connect over ethernet, you must be plugged into port 3 (on N600 or 750), and you have to give yourself a 172.30.0.10 IP like you did.

Loading from 172.22.0.1 or from https:// (which it looks like you are using in your screenshot?) will not work. But I think both of these issues were resolved a couple of days ago in the fix to https://github.com/sudomesh/peoplesopen-dash/issues/1.

I don't think anyone has rebuilt the firmware yet. We should put this fix in a patch.

bennlich, 2018-06-26:
> Update: was able to log into 172.30.0.1 after waiting patiently for it to get an IP address!

@jnny where did the IP address come from?

jnny, 2018-06-26:
I couldn't connect without ethernet until I figured out the password to the private net, and then.wasn't patient enough the first go-around.

Re: 172.22.0.1 - update to Walkthrough is loaded in my other comp's browser (next after coffee..). With images! .

Juul, 2018-07-11:
I think I found out why this is happening. The previous `/opt/mesh/pw_reset` script was actually deleting the entire `/etc/shadow` file at midnight (not 24 hours later, just the next time the router is on during midnight) and the `/etc/shadow` file also contains the password hash for the dashboard admin login so it is impossible to log in after that happens. I proposed a fix to both this issue and issue #142 in [this comment](https://github.com/sudomesh/sudowrt-firmware/issues/142#issuecomment-404048051) but it needs testing.

paidforby, 2018-08-06:
As suggested by @Juul, we think this is caused by the password reset confusion (i.e. mostly lazy coding/communication on my part). We corrected this issue in #142, I still have some reservations about the dashboard (which maybe addressed soon by issues like https://github.com/sudomesh/peoplesopen-dash/issues/1 and https://github.com/sudomesh/peoplesopen-dash/issues/5) and have noticed some idiosyncrasies with the mapping of ports to IPs (like why should port 3 hand out 172.22.0.1/24 addresses before configuring, but afterwards handouts 172.30.0.1/24 addresses) and, of course, the inaccuracies of the dashboard's port displays as described in https://github.com/sudomesh/peoplesopen-dash/issues/3.

I vote for leaving this issue open, as a way of linking progress on the dashboard issues back to the firmware.

bennlich, 2019-05-05:
closing this because you **can** log into the dash on 0.3.0, and pw_reset change suggested by @Juul was implemented. feel free to reopen if I'm missing something.

closed 2019-05-05 by bennlich