sudomesh/sudowrt-firmware #151: understand/rewrite meshrouting script

paidforby, 2018-10-10:
One part of the firmware that is still a mystery to me is the meshrouting script. It probably should be rewritten as a UCI config (using the `firewall` and `network` configs?), but first we need to understand the routing rules it is setting. This issue should be used to document those rules.

eenblam, 2018-10-10:
For reference:

[files/opt/mesh/templates/etc/init.d/meshrouting](https://github.com/sudomesh/sudowrt-firmware/blob/master/files/opt/mesh/templates/etc/init.d/meshrouting)

Shell variables imported by /etc/init.d/meshrouting and /etc/udhcp.user:
[files/opt/mesh/templates/etc/sudomesh/home_node](https://github.com/sudomesh/sudowrt-firmware/blob/master/files/opt/mesh/templates/etc/sudomesh/home_node)

A few more defined for MyNet devices:
[files/opt/mesh/templates](https://github.com/sudomesh/sudowrt-firmware/blob/master/files)/opt/mesh/templates/etc/sudomesh/mynet_n

paidforby, 2018-10-19:
brain dump regarding current VLAN, interface, bridge setup,  

On N600 home nodes VLANs are tied to physical ports. Instead of VLANs, we should describe the `open` and `private` networks as being "separate interfaces" that are "firewalled from one another." If we look at [/etc/config/network](https://github.com/sudomesh/sudowrt-firmware/blob/master/files/opt/mesh/templates/etc/config/network) we see that there are six interfaces; `ext1mesh`, `ext2mesh`, `mesh2`, `mesh5`, `open`, and `priv`. `ext1mesh` and `ext2mesh` are the extender node interfaces tied to `eth0.1` (i.e. port 1) and `eth0.2` (i.e. port4), respectively, and are what allow two N600s to "mesh" wiredly (either over a plain ethernet cable or over a mixed ethernet/wireless link). `mesh2` and `mesh5` are the ad-hoc (2.4Ghz and 5Ghz) interfaces that allow two N600s to "mesh" wirelessly. `open` is the bridge interface that bridges the public wireless interfaces (`open2` and `open5`), `eth0.10` (i.e. port 2), and the the `l2tp` interface created by tunneldigger. `priv` is the bridge interface that bridges the private wireless interfaces (`priv2` and `priv5`) and `eth0.11` (i.e. port 3). The wireless interfaces are defined in [/etc/config/wireless](https://github.com/sudomesh/sudowrt-firmware/blob/master/files/opt/mesh/templates/etc/config/wireless) and are tied to either `radio0` (the 2.4Ghz radio) or `radio1` (the 5Ghz radio). All of these interfaces are then either allowed or blocked from forwarding packets to one another by [/etc/init.d/meshrouting](https://github.com/sudomesh/sudowrt-firmware/blob/master/files/opt/mesh/templates/etc/init.d/meshrouting).