sudomesh/sudowrt-firmware #154: pplsopen-admin access to mesh

Jnull, 2019-01-27:
Is the ssid pplsopen-admin suppose to have access to the mesh network? 

The last paragraph of https://sudoroom.org/wiki/Mesh/Technical_Overview#The_three_wifi_networks_and_IP_assignment says
> The private network does not limit bandwidth and provides access to both direct access to the Internet (if the node owner has hooked the node up to the Internet) and access to the mesh.

This seems to not be the case, can someone confirm this? if not we may need to update the documentation or update the firmware to be inline with the documentation.

bennlich, 2019-02-28:
Sorry to take a month to reply to this @Jnull! I was going to say no, to my knowledge, the admin SSID is not supposed to have access to the mesh.

The admin SSID is configured in [/etc/config/wireless](https://github.com/sudomesh/sudowrt-firmware/blob/master/files/opt/mesh/templates/etc/config/wireless#L102), and it points to the `priv` network, which is defined in [/etc/config/network](https://github.com/sudomesh/sudowrt-firmware/blob/master/files/opt/mesh/templates/etc/config/network#L116-L122).

The `priv` network is on a different subnet (172.30.0.1/16) than the mesh (100.64.0.1/10).

However, there is an iptables rule in [meshrouting](https://github.com/sudomesh/sudowrt-firmware/blob/master/files/opt/mesh/templates/etc/init.d/meshrouting#L38-L39) with the comment "# Enable IP masquerading for private net access to mesh". I've never witnessed this behavior working, but it does sound like someone intended it to work when they wrote that line.

@Juul or @max-b do you recall ever being able to access mesh IPs from the admin SSID?

@Jnull That wiki page is accurately marked stale. I'm going to take a stab at updating / rewriting the stale pieces.

Jnull, 2019-02-28:
@bennlich np, it's not a huge problem, someone mentioned it to me but i missed that it was marked stale, apologies.

closed 2019-02-28 by Jnull