sudomesh/sudowrt-firmware #159: build 3.0 images for n750 and tp-link routers

wrought, 2019-11-23:
assuming this needs to be edited https://github.com/sudomesh/sudowrt-firmware/blob/master/openwrt_config/arch_configs/ar71xx

then the firmware tested and bugs fixed from there

bennlich, 2019-11-23:
Yessssssss

paidforby, 2019-11-24:
x-post from mesh mailing list

> assuming this needs to be edited https://github.com/sudomesh/sudowrt-firmware/blob/master/openwrt_config/arch_configs/ar71xx

Yes, line 3 needs to become `CONFIG_TARGET_ar71xx_generic_MYNETN750=y

> then the firmware tested and bugs fixed from there

The bugs that need to be fixed will be in the [pre-autoconf etc files](https://github.com/sudomesh/sudowrt-firmware/tree/master/files/etc) and the [post-autoconf etc files](https://github.com/sudomesh/sudowrt-firmware/tree/master/files/opt/mesh/templates/etc). I believe the biggest difference has to do with the layout of the ports/vlans. You can also reference the legacy makenode repo https://github.com/sudomesh/makenode to see if there are any hints. For example, this [etc/config/network](https://github.com/sudomesh/makenode/blob/master/configs/ar71xx/home_nodes/mynet_n/n750/templates/files/etc/config/network) file looks important and it appears to be the only N750 specific file.

wrought, 2019-12-03:
I'm rolling up my sleeves on this, but not making too much progress yet. I see the port/vlan layout differences especially concerning the [switch configuration](https://oldwiki.archive.openwrt.org/doc/uci/network/switch). Wishing there was some good documentation concerning OpenWRT's handling of building multiple images for different devices on the same architecture. 

For instance, I added a config target line instead of replacing in `arch_configs/ar71xx` , as I would like to build both n600 and n750 images. Further, I'm not totally wrapping my head around how to resolve #157.

paidforby, 2019-12-04:
You cannot target two different devices. Not sure why. It will only accept the second (i.e. the last executed) target. This is based on experience, though I may be able to find a source that corroborates my theory.  

My idea for creating multiple images is to re-run the build for each target. If done correctly each build should only take a minute or two. 

I don't have a lot of suggestion for working with the ports, I do love [this diagram](https://oldwiki.archive.openwrt.org/_detail/oldwiki/openwrtdocs/asus-internals-default.png?id=doc%3Auci%3Anetwork%3Aswitch) that you linked to, though.

bennlich, 2019-12-15:
@wrought Have you looked at the makenode configs for the n750? We used to be able to configure sudowrt 0.2.x for those devices with makenode, so I think if we copy the n750 configs from there, it ought to work. I'm down to look at this with you sometime if it would help to have some company :)

bennlich, 2019-12-15:
These are the n750 configs from makenode: https://github.com/sudomesh/makenode/tree/master/configs/ar71xx/home_nodes/mynet_n/n750

paidforby, 2019-12-16:
You are correct @bennlich, and that directory only contains one configuration, `/etc/config/network`. I believe that you should be able to just drop in the n750 config from makenode as a replacement for the n600 config in autoconf here, https://github.com/sudomesh/sudowrt-firmware/blob/master/files/opt/mesh/templates/etc/config/network.

autoconf has not deviated from makenode that much yet, so the template configs are essentially interchangeable, though we should confirm that all the entries that need to be updated in the n750 network config match up `uci set network` commands in `autoconf` here https://github.com/sudomesh/sudowrt-firmware/blob/master/files/opt/mesh/autoconf#L74.

I still think autoconf needs to be refactored. This issue of handling different hardware is another good reason, in addition to the ones i listed in #153.

paidforby, 2019-12-20:
I've (re)discovered the main difference between n600 and n750 configs. It appears that the n600 has a `wan` interface that is designated as `'eth1'`. On the n750, the `wan` interface is designated as vlan 5 on the eth0 with `'eth0.5'`.

The main change needs to be made in two files. 
First, [files/etc/config/network](https://github.com/sudomesh/sudowrt-firmware/blob/master/files/etc/config/network). This is the pre-autoconf network config file, and needs to have this vlan config added
```
config switch_vlan
  option device 'switch0'
  option vlan '5'
  option ports '0t 5'
```
and the `wan` interface changed to the following,
```
config interface 'wan'
  option ifname 'eth0.5'
  option proto 'dhcp'
```
Similarly, [files/opt/mesh/templates/etc/config/network](https://github.com/sudomesh/sudowrt-firmware/blob/master/files/opt/mesh/templates/etc/config/network), the post-autoconf network config, needs the same lines changed/added.

paidforby, 2019-12-21:
I made the suggested change in [this commit to the n750 branch](https://github.com/sudomesh/sudowrt-firmware/commit/a48c858da9679cab198b8d9376df5fd301f242c5). Someone should build this and test it! 

I also have some ideas on how to integrate this change into master so that y'all can automatically build binaries for both n600s and n750s. My ideas are loosely related to #153 so I may make some commits to master in preparation for a merge (assuming that commit worked).

paidforby, 2019-12-22:
Ok, so now it is actually building for n750. I realized the main reason for the existence for `openwrt_rebuilder` is to rebuild this for the n750. The lines of particular interest are,
```
  local build_dir=${workdir}/${BUILD_DIR}/builder.${arch}
  local kconfig=${build_dir}/scripts/kconfig.pl
  local config_dir=${workdir}/openwrt_config/arch_configs
  local config=${build_dir}/config.${arch}
  $kconfig 'm+' "${config_dir}/generic" "${config_dir}/${arch}" > "${config}"
```
In `openwrt_build_configure`, the configurations are saved to `built_firmware/openwrt/config.ar71xx`, but the configurations being used are in `built_firmware/builder.ar71xx/config.ar71xx`. So `openwrt_rebuilder` saves them to `built_firmware/builder.ar71xx/config.ar71xx`. I

t's always been unclear to me why these two directories exist, why don't we just always build in `built_firmware/openwrt/`. I'm sure this was an intentional design choice, I see a note in `build_lib` where it makes hardlinks to the build directory,
```
  local build_dir=${workdir}/${BUILD_DIR}/builder.${arch}

  if [ ! -d "$build_dir" ]; then 
    # Make hardlinks to avoid space duplication but to still isolate builds
    cp -lr "${workdir}/${BUILD_DIR}/openwrt" "${build_dir}"
  fi
```
I'm not sure why we need to "avoid space duplication" and "still isolate builds"... who knows.

wrought, 2020-02-19:
Hi, thanks for your work on this @paidforby 

Coming back to this task -- so it seems from what I can tell you're saying the `n750` build is building but it has not been flashed and tested, is that correct? If so, I can run the build again and try to use the firmware it generates on the router I have. 

In any event, it seems like the task of supporting both images would require further modification of the build scripts, right? That was what I would like to help accomplish so that we can continue to flash n600 and n750 models until we run out. 

I also created issue #163 to track moving the configurations over to an openwrt package instead of maintaining the sudowrt-firmware pre-script scripts.

paidforby, 2020-02-21:
> Coming back to this task -- so it seems from what I can tell you're saying the n750 build is building but it has not been flashed and tested, is that correct? If so, I can run the build again and try to use the firmware it generates on the router I have.

So I believe I left this task at the following point: I built an n750 image. I flashed that image to an n750. It was apparently successful, but when the n750 boots it is not providing DHCP on any of the ports, nor can I ping by setting a static in the possible default ranges.

I'm not sure how much I debugged. I know that I did not have an ethernet to plug in and test the WAN port, not sure that would make much difference.

I think the key thing to play with is the etc/config/network file. But I could be wrong, I don't want to discourage you from following other leads.

HOWEVER, I think #163 is a much more worthy project to pursue. Maybe you could flash vanilla OpenWrt to the n750, install babeld and tunneldigger, and then figure out how to turn autoconf into a package.

wrought, 2020-02-26:
Aha, I had misread your earlier comment, I am going to attempt to build, flash, test, and fix any issues to hopefully get this branch working for n750 builds (for now). 

Otherwise, I plan to proceed with #163 at this point. In fact, we may have gotten some Microtik and other hardware to work with tonight.