sudomesh/sudowrt-firmware #160: update version of openwrt

wrought, 2019-12-11:
going to need to do a lot of testing and probably manage conflicts with packages / configurations.

wrought, 2019-12-11:
@Juul

paidforby, 2019-12-16:
agreed. honestly, upgrading to OpenWrt 18 from the current firmware will be a challenge. I highly recommend rebuilding the firmware from scratch with the idea of maintaining interoperability with old home nodes and exit nodes.

Really, peoplesopen nodes only need to do two things,
1. babel with home nodes
2. dig a tunnel to exit nodes

So why not rethink the rest of the configurations. There must be easier ways in the latest OpenWrt to do a lot of the interface, routing, and firewall configs we do for the old OpenWrt 15 (Chao Calmer) firmware. Plus, you need to find new hardware anyway, so some configurations may not even translate to that new hardware.

Maybe it's just me, but a full rebuild sounds more fun and more enlightening than wrestling with package and configuration conflicts in our old, crufty firmware.

wrought, 2020-02-19:
I am closing in favor of the path in #163

closed 2020-02-19 by wrought