sudomesh/sudowrt-firmware #163: Create sudowrt package

wrought, 2020-02-19:
After conversations with @Juul @mitar and others, it seems like it makes sense to move away from the sudowrt-firmware building process to utilizing the package system available with openwrt. 

Apparently the main issue had been the "extender node" feature which required some firmware modifications that would not have been possible as a package. However, this is not a desired aspect of the firmware moving forward anyhow. There may be other limitations or obstacles as well, but overall, this seems to be a superior option in order to support many more hardware platforms, decrease developer overhead, and make maintenance more straight-forward.

paidforby, 2020-02-21:
This is a great idea. The autoconf script does most of the magic need to turn a router into a people's open node. I bet if you took vanilla OpenWrt, installed a few packages (namely babled and tunneldigger) and then ran autoconf (with maybe a few modifications) you would be halfway to your goal.

sunjam, 2020-03-21:
It would be great to run this on my home router!  Can't wait!

david4096, 2022-03-19:
Anybody still working on this?

paidforby, 2022-03-29:
@david4096 Not that I know of. I'm not sure there any exit nodes still running or if the node database (secrets.peoplesopen.net) is still working? Reviving this project would take some serious doing. I recommend taking a look at LibreMesh's lime packages repos, https://github.com/libremesh/lime-packages, if you want openwrt packages that turn a normal router into a mesh router. That is an infinitely more active project with some awesome folks working on it.